Ultimaker/Cura #3026: Three a head, 3 color printing how to set the remaining material removal nozzle. Cura support dual color removal

xc3dmaker, 2017-12-21:
Application Version:

Platform:

Qt:

PyQt:

Display Driver:

Steps to Reproduce:

Actual Results:

Expected results:

Additional Information:

ChrisTerBeke, 2017-12-21:
I'm not sure what your question is...

you want to set the nozzle for support material?

xc3dmaker, 2017-12-21:
I'd like to set up 3 color prints, three feed motors, a nozzle. The nozzle will have residual red material after the nozzle 1 red, and the nozzle 2 and 3 need to be removed from the nozzle residual material. Otherwise, print the sprinkler 2 color is not pure.

ChrisTerBeke, 2017-12-21:
That's something the machine's firmware should handle, as Cura doesn't know wether 3 extruders are attached to the same nozzle or not. We actually have a machine listed (Builder Premium) that has this functionality and uses firmware to correctly implement the marlin/reprap protocol to make it work. Maybe you can try setting up 3 extruders in Cura (custom FDM printer) and then use a larger prime tower?

xc3dmaker, 2017-12-21:
I have tested it. Cura supports double color. Three nozzles are printed on the same floor tower and failed when printing.

xc3dmaker, 2017-12-21:
We have no problem with double color.

ChrisTerBeke, 2017-12-21:
This is a duplicate of https://github.com/Ultimaker/Cura/issues/1936 and https://github.com/Ultimaker/CuraEngine/issues/588 (the latter of which might have been using the same diamond extruder).

closed 2017-12-21 by ChrisTerBeke

xc3dmaker, 2017-12-21:
> Cura doesn't support wipe towers for more than 2 extruders indeed. We have something in our backlog to fix this (a tower per extruder, different shapes, etc) but I don't think this will be implemented anytime soon as Ultimaker printers have max 2 extruders.

How much time and money will take if it is customized and how much is the cost?

ChrisTerBeke, 2017-12-21:
@Appesteijn

Appesteijn, 2017-12-21:
You could give each nozzle it's own prime tower, but we (Ultimaker) are not going to build that in the near future. Mainly because we don't have a machine with more that two nozzles yet.