Ultimaker/Cura #3027: When double color printing, how to cancel standby temperature, printing often needs to wait for temperature to rise to the specified temperature.

xc3dmaker, 2017-12-21:
Application Version:

Platform:

Qt:

PyQt:

Display Driver:

Steps to Reproduce:

Actual Results:

Expected results:

Additional Information:

ChrisTerBeke, 2017-12-21:
You can change the standby temperature in the custom settings. We recommend against it though as the current settings have been tested thoroughly.

closed 2017-12-21 by ChrisTerBeke

xc3dmaker, 2017-12-21:
The standby temperature is set at 210 degrees, and it will continue to wait and wait

xc3dmaker, 2017-12-21:
The printing temperature is in accordance with the standby temperature

ChrisTerBeke, 2017-12-21:
Which Cura version are you using?

xc3dmaker, 2017-12-21:
cura 3.1

ChrisTerBeke, 2017-12-21:
Can you send a project file so we can debug this? You can export your whole workspace under file -> save as project. Then upload that file here (not via email but the GitHub interface).

Ghostkeeper, 2018-01-15:
Attachments to e-mails that reply to Github notifications won't work. You need to actually go on the Github website and upload it there.

Ghostkeeper, 2018-01-15:
If you want to eliminate the temperature changes, you need to remember to adjust the "Initial Printing Temperature" and "Final Printing Temperature" as well. Otherwise it will still cool.

It might be worth your while to improve the heating/cooling speed estimates for your printer type in the definition file, if you are technically inclined.