Ultimaker/Cura #3034: Keep Post Processing Plugin enabled?

Tomblarom, 2017-12-22:
Hey. If I set some scripts to run over my gcode and close cure, they won't be saved.. Is it a bug? I want to use the post processing settings for all of my prints.

Thank you,
Tombla

ChrisTerBeke, 2017-12-29:
They're currently not saved by design as you might forget about them and get unexpected printing results. I think the discussion whether to keep this behaviour or change has been held several times and so far we kept it the same.

Ghostkeeper, 2018-01-16:
Well, that's still up for discussion.

Duplicate of https://github.com/Ultimaker/Cura/issues/2796, https://github.com/nallath/PostProcessingPlugin/issues/46 and https://github.com/Ultimaker/Cura/issues/2854.

closed 2018-01-16 by Ghostkeeper