Ultimaker/Cura #3037: How to resolve: undefined reference to `pthread_create' while running cmake command in cura-build

veilupt, 2017-12-23:
I have an ubuntu machine to run cura-build from the following doc.

Refer: https://github.com/Ultimaker/cura-build#ubuntulinux

My gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.5)

while running cmake command

I am getting this error,

```
~/ cura-build/build$ cmake …
cmake: /usr/lib64/tableausdk/libcurl.so.4: no version information available (required by cmake)
CMake Error at CMakeLists.txt:24 (get_filename_component):
get_filename_component unknown component CACHE

– Configuring incomplete, errors occurred!
See also “/cura-build/CMakeFiles/CMakeOutput.log”.
See also “/cura-build/CMakeFiles/CMakeError.log”.
```

In the CMakeError.log,

```
CMakeFiles/cmTC_2f3cc.dir/CheckSymbolExists.c.o: In function main': CheckSymbolExists.c:(.text+0x16): undefined reference topthread_create’
collect2: error: ld returned 1 exit status
CMakeFiles/cmTC_2f3cc.dir/build.make:97: recipe for target ‘cmTC_2f3cc’ failed
make[1]: *** [cmTC_2f3cc] Error 1
make[1]: Leaving directory '/cura-build/CMakeFiles/CMakeTmp’
Makefile:126: recipe for target ‘cmTC_2f3cc/fast’ failed
make: *** [cmTC_2f3cc/fast] Error 2

File /cura-build/CMakeFiles/CMakeTmp/CheckSymbolExists.c:
/* */
#include <pthread.h>

int main(int argc, char** argv)
{
(void)argv;
#ifndef pthread_create
return ((int*)(&pthread_create))[argc];
#else
(void)argc;
return 0;
#endif
}

```
And also tried to compile gcc,

```
$ gcc -pthread
gcc: fatal error: no input files
compilation terminated.
```

veilupt, 2017-12-26:
Now I resolved cmake issue by setting the CMAKE_PREFIX_PATH. 
But I can't able to run cura-build because of the CuraEngine seems to be an error. 
How to resolve this? 
`/usr/local/lib/libArcus.so.1.1.0: undefined reference to `google::protobuf::io::CodedInputStream::SetTotalBytesLimit(‌​int)' collect2: error: ld returned 1 exit status CMakeFiles/CuraEngine.dir/build.make:98: recipe for target 'CuraEngine' failed CMakeFiles/Makefile2:142: recipe for target 'CMakeFiles/CuraEngine.dir/all' failed`

Ghostkeeper, 2018-01-15:
Was libArcus compiled using the same compiler as CuraEngine?

veilupt, 2018-01-18:
@Ghostkeeper Yes, Now modified the path which is pointing to the correct Protobuf I used in the CuraEngine. Now I can able to run. Thanks

closed 2018-01-18 by veilupt