Ultimaker/Cura #3038: Fresh Install Fedora 27 w/ Intel HD Graphics 620 Segfault

bghost4, 2017-12-23:
**Application Version**
Cura -3.1.0.AppImage from Ultimaker Site

**Platform**
Fedora Core 27
$uname -a
Linux flippyfloppy 4.13.9-300.fc27.x86_64 #1 SMP Mon Oct 23 13:41:58 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

$ cat /etc/fedora-release 
Fedora release 27 (Twenty Seven)

**Display Driver**
Intel HD Graphics 620
$ lspci | grep VGA
00:02.0 VGA compatible controller: Intel Corporation HD Graphics 620 (rev 02)
00:13.0 Non-VGA unclassified device: Intel Corporation Sunrise Point-LP Integrated Sensor Hub (rev 21)

**Steps to Reproduce**
Run Cura : ./Cura-3.1.0.AppImage

**Actual Results**
./Cura-3.1.0.AppImage 
Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 72: non-double matrix element
Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 72: non-double matrix element
Fontconfig warning: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 80: saw unknown, expected number
Fontconfig warning: "/etc/fonts/conf.d/65-0-lohit-bengali.conf", line 32: unknown element "langset"
Fontconfig warning: "/etc/fonts/conf.d/69-gnu-free-sans.conf", line 24: unknown element "langset"
Fontconfig warning: "/etc/fonts/conf.d/69-gnu-free-serif.conf", line 24: unknown element "langset"
QObject::connect: Parentheses expected, signal MainWindow::
QObject::connect: Parentheses expected, signal MainWindow::
QObject::connect: Parentheses expected, signal MainWindow::
QObject::connect: Parentheses expected, signal MainWindow::
file:///tmp/.mount_dR6dqB/usr/bin/resources/themes/cura-light/styles.qml:197:17: QML Item: Binding loop detected for property "width"
qml: TableViewSelection: index out of range
qml: TableViewSelection: index out of range
./bin//cura.sh: line 12: 22386 Segmentation fault      (core dumped) cura $@

**Expected results**
Software to show User Interface without crashing

**Additional Information**
Fresh Install of Fedora 27 This morning, Did all Updates.

LipuFei, 2018-01-03:
Seems to be the same issue as https://github.com/Ultimaker/Cura/issues/3023

Ghostkeeper, 2018-01-22:
Yes, this is a duplicate.

It was solved by https://github.com/Ultimaker/Cura/pull/2903 .

closed 2018-01-22 by Ghostkeeper