Ultimaker/Cura #3039: USB support for MakerBot (compatible) printers?

stuartpb, 2017-12-23:
I'm using a Qidi Tech I (Replicator clone running Sailfish v78), and I can't connect to it from Cura 3.1. I've stopped the MakerBot Conveyor Service that was denying access to COM7 (the printer's port), but even with that turned off, Cura fails to connect to / recognize the printer at startup (it does cause the printer to play its "printer ready" chime a couple of times, though):

```
2017-12-23 14:28:05,920 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin FileLogger
2017-12-23 14:28:06,288 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin FirmwareUpdateChecker
2017-12-23 14:28:06,357 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin GCodeProfileReader
2017-12-23 14:28:06,396 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin GCodeReader
2017-12-23 14:28:06,434 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin GCodeWriter
2017-12-23 14:28:06,542 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin ImageReader
2017-12-23 14:28:06,584 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin LegacyProfileReader
2017-12-23 14:28:06,644 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin LocalFileOutputDevice
2017-12-23 14:28:06,697 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin MachineSettingsAction
2017-12-23 14:28:06,977 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin OctoPrintPlugin
2017-12-23 14:28:07,133 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin PerObjectSettingsTool
2017-12-23 14:28:07,192 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin PluginBrowser
2017-12-23 14:28:07,228 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin PostProcessingPlugin
2017-12-23 14:28:07,339 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin RemovableDriveOutputDevice
2017-12-23 14:28:07,493 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin SimulationView
2017-12-23 14:28:07,578 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin SliceInfoPlugin
2017-12-23 14:28:07,617 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin SolidView
2017-12-23 14:28:07,674 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin CameraTool
2017-12-23 14:28:07,752 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin MirrorTool
2017-12-23 14:28:07,814 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin RotateTool
2017-12-23 14:28:07,891 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin ScaleTool
2017-12-23 14:28:07,929 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin SelectionTool
2017-12-23 14:28:07,999 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin TranslateTool
2017-12-23 14:28:08,273 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin UltimakerMachineActions
2017-12-23 14:28:08,491 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin UM3NetworkPrinting
2017-12-23 14:28:08,546 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin UpdateChecker
2017-12-23 14:28:08,548 - INFO - UpdateChecker.UpdateCheckerJob.run [34]: Checking for new version of cura
2017-12-23 14:28:08,696 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin USBPrinting
2017-12-23 14:28:08,749 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin UserAgreementPlugin
2017-12-23 14:28:08,900 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade21to22
2017-12-23 14:28:08,965 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade22to24
2017-12-23 14:28:09,023 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade25to26
2017-12-23 14:28:09,084 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade26to27
2017-12-23 14:28:09,153 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade27to30
2017-12-23 14:28:09,215 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade30to31
2017-12-23 14:28:09,241 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin SimpleView
2017-12-23 14:28:09,312 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin X3DReader
2017-12-23 14:28:09,402 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin XmlMaterialProfile
2017-12-23 14:28:09,449 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin XRayView
2017-12-23 14:28:09,497 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [312]: Attempting to connect to COM5
2017-12-23 14:28:09,498 - INFO - UM.VersionUpgradeManager.upgrade [112]: Looking for old configuration files to upgrade.
2017-12-23 14:28:09,499 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [312]: Attempting to connect to COM7
2017-12-23 14:28:09,612 - DEBUG - UM.Backend.Backend._logSocketState [174]: Socket state changed to Listening
2017-12-23 14:28:09,756 - INFO - UM.Backend.Backend.startEngine [75]: Started engine process: C:\Program Files\Ultimaker Cura 3.1\CuraEngine.exe
2017-12-23 14:28:09,758 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] Calling engine with: ['C:\\Program Files\\Ultimaker Cura 3.1\\CuraEngine.exe', 'connect', '127.0.0.1:49674', '-j', 'C:\\Program Files\\Ultimaker Cura 3.1\\resources\\definitions\\fdmprinter.def.json', '']
2017-12-23 14:28:09,761 - WARNING - UM.OutputDevice.OutputDeviceManager.removeOutputDevice [126]: Could not find output device with id COM5 to remove
2017-12-23 14:28:09,762 - WARNING - UM.OutputDevice.OutputDeviceManager.removeOutputDevice [126]: Could not find output device with id COM7 to remove
2017-12-23 14:28:10,194 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] 
2017-12-23 14:28:10,237 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] Cura_SteamEngine version DEV
2017-12-23 14:28:10,239 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] Copyright (C) 2017 Ultimaker
2017-12-23 14:28:10,241 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] 
2017-12-23 14:28:10,243 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] This program is free software: you can redistribute it and/or modify
2017-12-23 14:28:10,247 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] it under the terms of the GNU Affero General Public License as published by
2017-12-23 14:28:10,249 - DEBUG - UM.Backend.Backend._logSocketState [178]: Socket state changed to Connected
2017-12-23 14:28:10,251 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] the Free Software Foundation, either version 3 of the License, or
2017-12-23 14:28:10,254 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] (at your option) any later version.
2017-12-23 14:28:10,255 - DEBUG - UM.Backend.Backend._onSocketStateChanged [168]: Backend connected on port 49674
2017-12-23 14:28:10,257 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] 
2017-12-23 14:28:10,260 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] This program is distributed in the hope that it will be useful,
2017-12-23 14:28:10,262 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] but WITHOUT ANY WARRANTY; without even the implied warranty of
2017-12-23 14:28:10,264 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
2017-12-23 14:28:10,267 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] GNU Affero General Public License for more details.
2017-12-23 14:28:10,269 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] 
2017-12-23 14:28:10,271 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] You should have received a copy of the GNU Affero General Public License
2017-12-23 14:28:10,273 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] along with this program.  If not, see <http://www.gnu.org/licenses/>.
2017-12-23 14:28:11,493 - INFO - USBPrinting.USBPrinterOutputDevice._connect [320]: Could not establish connection on COM5: 'Timeout'. Device is not arduino based.
2017-12-23 14:28:11,499 - INFO - USBPrinting.USBPrinterOutputDevice._connect [320]: Could not establish connection on COM7: 'Timeout'. Device is not arduino based.
[... 33 lines cut ...]
2017-12-23 14:28:38,117 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [C:\Program Files\Ultimaker Cura 3.1\resources\shaders\composite.shader]...
2017-12-23 14:28:38,128 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [C:\Program Files\Ultimaker Cura 3.1\resources\shaders\overhang.shader]...
[... cut ...]
2017-12-23 14:29:14,280 - DEBUG - USBPrinting.USBPrinterOutputDevice.close [385]: PrinterConnection.close: cannot join current thread (expected)
2017-12-23 14:29:14,283 - WARNING - UM.OutputDevice.OutputDeviceManager.removeOutputDevice [126]: Could not find output device with id COM7 to remove
```

Is there something more that could be done to make this connection work? The ability to fire off prints directly from the desktop (without setting up an RPi with OctoPrint) is the one thing currently keeping me on Makerbot Desktop.

ChrisTerBeke, 2017-12-29:
Probably not happening by Cura core team as Makerbot is sort of a direct competitor. Any 3rd party contributions are welcome of course. There is an [X3G writer plugin](https://github.com/Ghostkeeper/X3GWriter) available already though, but I'm not sure how newer Makerbot machines work with that or if the comm. between printer and software is encoded or something.

Ghostkeeper, 2018-01-16:
USB printing support is ready in the next version of the X3G writer. But the next version is waiting on https://github.com/Ultimaker/Cura/pull/2960 to be merged first.

rpavlik, 2018-02-16:
oh, I didn't realize there was a dev branch...  GPX (used to make x3g files) can run in a transparent-translator mode, but that only works on Linux. (It's how octoprint can drive x3g printers over USB, though it has its hiccups)

ianpaschal, 2018-03-27:
@stuartpb Can you confirm if you're still having this issue now that the aforementioned fixes have been merged?

nallath, 2019-01-07:
Closing due to inactivity

closed 2019-01-07 by nallath