Ultimaker/Cura #3040: Pause at height CR-10 Unexpected Results

miniquandt, 2017-12-24:
I have a CR-10 and for the most part once Cura is set up I have not had any issue with it. But for some reason when I try to "Pause at Height" it never actually pauses at the specified height.

Background:
I just set up an OctoPi, and I am able to view the G-Code before I can print the file. I spent about an 1hr and 30 minutes trying to get the G-Code to pause at specific heights so that I can exchange the filament to get a multi colored print. I ended up creating 36 different files just trying to fine tune and get the printer to pause at the correct height.

What I did:
I have a file approximately 7mm tall, and I wanted to stop at around the 3mm, 4mm, 4.75mm, 5.25mm, 6.5mm. So initially I entered those values and I got wacky values back. (something like 3.6, 4.7, 5.2, 6.7, 7.8, 8.4) some of which were above my model. So then when I tried to fine tune it, I started off at a Pause at height 1, and got back 1.2. Then went pause at height 2 and got back 2.3. I ended up getting 3 out of the 5 perfectly aligned with the print, and 1 close enough to print. The last one would not get anywhere close. Either it was too low (like between values 2-3) or above the print.

My last settings are Pause at height: 1.75, 2.65, 4.25, 4.252. and they matched up at about 3.1mm, 4.1, 5.25, 6.3. I believe that the last value I started off at 4.26 which matched to 6.3, I shrunk it down trying to get the last pause in.

I still have the 36 g-code files and would post them if you want them. But I dont know if they will help. FYI I will be out of town from Dec 29-Jan 9. I will try to check here before I leave to see if you want me to post some of them. 

**Application Version**

**Platform**

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

miniquandt, 2017-12-24:
I may have just spent 2 hours of my life over a problem that isn't your fault. It might have been Octoprint which was displaying the Pause Height. I just ran the code which should have stopped at the correct height with the wrong values and either the print crashed, or it paused and the octopi crashed. I might be closing this myself. Sorry if that is the case

miniquandt, 2017-12-24:
Yeah, I tried printing and the pause at height was completly wrong from what Octoprint said it should be. I dont know if it is Cura, or if I just got really unlucky with two prints. Im gonna close it because I cant even explain what happened.

closed 2017-12-24 by miniquandt

Ghostkeeper, 2018-01-15:
If you save it to a file, open that file with a text editor and find that the pause is at the wrong position, then we'll consider this a Cura bug ;)