Ultimaker/Cura #3041: sPECIAL mODE ONE AT A TIME

twitteking, 2017-12-24:
Version one

This is not as much a bug in the stability, but a functionality "bug"

The main variable Cura takes into account is the gantry height currently, in preventing colliding with printed pieces.  BUt there is more exploitation!

I argue we can actually print above the gantry height as along as the the print priority for piece are based on position of the model on the diagonal far corner away from auto-home first, since the gantry wont go back to that position if you can imagine what i am saying. thank you!

Importance
i am recently working on some things that i need print 30+ copy of small parts ( 7 to 20 mins printing), i found the one at a time setting relatively convenient seem i can do post print processing on the first piece while the second one is printing, also it is more risk free from print accident like power-outage or layer misalignment. of course the best part is i dont need to keep manually initiate print. thats why i think many people with bigger build volume would find it useful!

**Actual Results**

**Expected results**

**Additional Information**

Ghostkeeper, 2018-01-15:
Not all mechanical arrangements have bars along the X/Y axes. Some are H-bar gantry systems, Scara, or have a fixed arm coming from an arbitrary direction and then a moving bed. A diagonal line of pillars is not always possible then.

Having a full plane "roof" as a model for the gantry system is a simplification that's usually fine.

Ghostkeeper, 2018-01-29:
That depends entirely on the order it is printed in though. The order in which it is printed is determined in a way that minimises the collision area of the print head. If the order is changed (such that the highest object is printed last) then the print head might collide with an object that's next to it when the print head moves to print the last item.

ianpaschal, 2018-03-05:
I think it's highly unlikely that we'll move away from the ceiling plane set-up.

closed 2018-03-05 by ianpaschal