Ultimaker/Cura #3043: Won't save Extruder Start and End Gcode in machine settings

tkuan, 2017-12-24:
Cura 3.1
MacOS 10.10.5

Try to add Extruder Start and End Gcode in the machine settings, but it will disappear after relaunch the program, and the codes are not inserted into the sliced the gcode.

ndladis, 2017-12-24:
Cura 3.1.0
Linux  OpenSuse 13.2

As it looks like I have the same problem. Whenever I try to save start and end gcode on an extruder it totally disregard it as soon as I close the printer settings. 
The machine I am trying to configure is a custom FDM printer dual extrusion.

seblacasse1986, 2017-12-25:
I have the same problem. Is it possible to modify a file instead as a work around ??

tkuan, 2017-12-25:
I have no choice but moved back to version 3.0.4

ChrisTerBeke, 2017-12-29:
Duplicate of #1296. Please search through old issue before opening a new one.

closed 2017-12-29 by ChrisTerBeke