Ultimaker/Cura #3048: wall start underextrusion

ChokHaz, 2017-12-26:
**Application Version**
 cura 3.1.0
I´m having problems of underextrusion at the start of every line of the prints, it is not normal underextrusion because it only happens after a long movement when a retraction is made, also it doesn´t happend in every layer but when it happens always looks the same. I couldn´t find a solution for this online, I tried changing different settings but it didn´t work, in the slic3r prusa edition at the retraction settings I found an option that may work called ´´extra lenght on restart´´ that pushes a little bit of material after a retraction, but I couldn´t find a similar option in cura so I don´t know what to do, I have never seens this issue before and I don´t remember having it with the previous cura versions. If you are going to help me please don´t talk about editing gcode or thinks like that because I don´t understand that kind of things.
I´m using a prusa i3 mk2s and esun pla+ filament but it also happens with other filaments.
[image]
[image]
[image]
[image]

smartavionics, 2017-12-26:
The Cura equivalent setting is Material -> Retraction Extra Prime Amount.

CCS86, 2017-12-26:
What is your retraction distance and speed for retract and prime?

Also, what are your print speeds (wall / infill)?

I think the likely culprit is skipping steps on the extruder while trying to prime the nozzle.

smartavionics, 2017-12-26:
Hi, I have a bowden extruder so the distance is 5, retract at 50mm/s, prime at 30mm/s, extra prime amount 0. I use various speeds but none are very fast as I prefer quality over speed. The part I am printing right now uses just 25mm/s for the walls and 50 for the infill. Sometimes I use 50/60 for the walls, just depends what I am printing.

ChokHaz, 2017-12-28:
After some tests trying with different retraction extra prime amount values I still have the same problem, the only thing that this setting did was make thin parts and supports full of blobs and causing the supports to fuse with the parts.

ChokHaz, 2017-12-28:
Retraction distance is 0.8mm and speed 35mm/s
I´m printing at 80mm/s at all the walls and infill and travel 120mm/s

ChokHaz, 2017-12-29:
What I think is the most weird, is that even when I disable the retraction the underextrusion still happens.

jackha, 2018-01-02:
Some ideas to check:
- Does your print have big speed changes? Switching from a low speed instantly to a high speed could potentially cause underextrusion right after speedup, because the bowden tube reacts slowly.
- Does it move a long distance before printing the underextruded parts?

nallath, 2018-01-03:
Big speed changes are mostly an issue with bowden style printers. Most non ultimaker machines have direct drives, so they don't suffer that much from this.

ianpaschal, 2018-03-26:
Haven't heard back from OP in months. Closing.

closed 2018-03-26 by ianpaschal