Ultimaker/Cura #3049: Improve Combing

thesec, 2017-12-26:
To further improve combing I would like to suggest to add the following settings to it:

- combing speed
- Z-Hop with combing

CCS86, 2017-12-26:
Combing is there to avoid a retracted travel move by leaving the nozzle in contact with the print to prevent oozing. Z hop would defeat the purpose.

thesec, 2017-12-26:
It might, it does not per se.
If it defeats oozing depends on more than "the Z-Hop". That option would still be appreciated.

CCS86, 2017-12-27:
What are you trying to say?

There are 3 types of travel moves:

-combing: where the nozzle stays against the print (no retract, no lift)
-retracted travel (no lift)
-retracted travel (with z-hop)

By adding a z-hop without a retract, there will be oozing. This is not desirable.

thesec, 2017-12-27:
There must not be oozing, depending on at least temperature, nozzle diameter and length of the move ... not only a retract happening or not.

By adding a slight z-hop one can prevent the nozzle to collide with printed parts which is desirable.

Adding the option(!) is desired; retraction might be added as well if found helpful.
If you do not want to use either, leave it off.

CCS86, 2017-12-27:
I still don't understand what you are asking for. It sounds like a combination that already exists.

-Disable combing
-Either disable retraction, or set a minimum travel distance large enough to allow the non-retracted moves you are trying to get
-Enable z-hop over printed parts

thesec, 2017-12-28:
Neither will leave the travel moves above printed parts.

Z-Hop aside, request for the combing speed seems crystal clear.

Ghostkeeper, 2018-01-16:
I'm agreeing with CCS86 on this argument. If you do a z-hop you're going over everything, so combing will be unnecessary because you're not colliding with anything anyway. If the normal combing still hits your print, you might need to increase the Travel Avoid Distance.

Combing doesn't really prevent oozing since you're combing through sparse infill sometimes. The ooze happens on the inside of the model, so you'll still lose some material but it won't be visible on the outside bar underextrusion.

The request for combing speed is clear. But why is it necessary to have a different speed for combing than the speed for other travel?

thesec, 2018-01-16:
There's no such thing as "Travel Avoid Distance"? At least the search does not reveal it, nor do I see it when looking through the settings.

The ooze happening on the model might cause the nozzle to get clogged upon infill, larger clogs might drop elsewhere on the model and cause openings in walls, wrong colors showing up etc.

While the travel speed might be (very) slow, the combing speed might be (very) fast as low speeds are likelier to have the above happening with it.

CCS86, 2018-01-16:
Why would you ever want slow travel speeds? You want to travel as fast as your machine is mechanically capable.

thesec, 2018-01-16:
Come on ... this cannot be serious!

CCS86, 2018-01-16:
Answer the question. Enlighten me.

The only possible reason I can think of for to travel slower, is to reduce resonance.

thesec, 2018-01-17:
Get some printing experience, troll some other thread.

CCS86, 2018-01-17:
No thanks. I saw your post before you edited it. I'm not going to waste any more of my time on your half baked ideas. I was trying to help you talk through your functional requirements, but you can't even make a case for **why** you want, what you say you do. Only one dev has chimed in, and they agree with my thoughts. I wonder which one of us is actually "lacking 3d printing experience". Later!

thesec, 2018-01-17:
Thanks for no longer wasting time, indeed.

ChrisTerBeke, 2018-01-17:
We're not gonna remove any of these comments since we want to be transparent, but please keep it civil here folks. Thanks!

Ghostkeeper, 2018-02-05:
The "Travel Avoid Distance" setting does exist. It's in the Travel category.

But it's still not clear why there needs to be a difference between combing travel moves and non-combing travel modes. Maybe your acceleration setting is too high?

thesec, 2018-02-07:
There's no distance setting in my Cura 3.1.0 instance, even with all settings enabled.
[image]

thesec, 2018-02-11:
Got the "Travel Avoid Distance" to show ... but this is working on X/Y only.
Allowing to avoid a distance on Z would allow for straighter moves and a speedier finish and (a) without clogging the nozzle or the current layer - which combing the model does unless you set a very high travel avoid distance and hence make the combing obsolete - and (b) while still avoiding the need to retract.

As a bonus one could think of a "landing zone", such as with coasting where the nozzle is not dropped back on the model on the target coordinates but gets an adjustable length to land back on the current layer.

I still see this addition a major improvement :)

Ghostkeeper, 2018-03-05:
> Allowing to avoid a distance on Z would allow for straighter moves and a speedier finish and (a) without clogging the nozzle or the current layer - which combing the model does unless you set a very [...]

This is exactly what Z-hop does. Straight lines, no combing any more, no retraction needed, and it makes sure to land inside the infill.

thesec, 2018-03-05:
And Z-hop does stay within the model's dimensions, same as with combing?

thesec, 2018-03-10:
But Z-Hop does not keep the nozzle in the model dimensions?

thesec, 2018-04-11:
Aside Z-hop being able to be set to other than 2 mm there is atm no option to combine combing - aka keeping the nozzle within the model dimensions - and lifting the nozzle. Adjusting the speed for combing is still thought of helpful ...

... moving in circles ... :/

I'll leave it up to you to decide on either suggestion.

Ghostkeeper, 2018-05-03:
As for comb speed vs. travel speed: When you don't comb, you're also moving the nozzle through the already printed areas so you could just as well hit a blob in the filament at full speed. There are really no differences except that combing moves might have corners in them and hit the walls less often.

I still don't understand why you'd need to comb while you're up in the air far above your model :confused: If you'll leave it up to me, so far it's a no, I don't see the advantage to this that you're trying to explain.

thesec, 2018-05-04:
When not combing I have retract at layer change and before walls as options and can adjust the speed to my needs, separately for infill etc.

If combing is no different than what you get with all other already - as was claimed multiple times by now - I do not see any benefit in combing, other than cluttering the UI. If it is indeed different I do not understand where (can someone add an illustrating image perhaps?) and do not get why it should not make any sense at all to allow this movement mode the same options others already have and consistency being a very basic pattern of good UX.

Ghostkeeper, 2018-06-04:
I'll try to explain my point of view differently again then:

Combing:
- \+ Hit fewer walls
- \- Make horizontal detours, longer travel moves

Z-hop:
- \+ Hit no walls
- \- Make a detour via the Z axis

Combing + Z-hop:
- \+ Hit no walls
- \- Make horizontal detours, longer travel moves
- \- Make a detour via the Z axis

This is why it makes no sense to me to enable combing when you have Z-hop. It's strictly worse than just Z-hop.

ianpaschal, 2018-07-02:
Since this hasn't seen any action in 4 weeks and it's been fairly clear since the beginning of the thread that we can't really justify development effort or cluttering the code, I'm going to close it lest we forget about it for months. I think @Ghostkeeper's answer is pretty clear.

closed 2018-07-02 by ianpaschal

darkracer125, 2021-10-30:
extremely old.  but since it's the first google result on "combing speed" imma just comment here anyway.  

forget about the z-hop.  can we focus on the combing speed for a minute here?  

i want to be able to force it to comb with a fast speed.  wich it is currently not doing on the first layer. (cura 4.5) 
i have a model with screw holes in it. (a mounting bracket) 
and the problem is,  that it starts on one hole,  then combs (since it's not retracting) to the hole right next to it.  but it doesn't do it at the travel speed.    if i have to guess it's about 45 m/s  
first layer print speed was down to 15m/s (because i was having this problem and i thought it was the new spool) 
travel speed is 150m/s 

in any case the problem is that in the slower combing moves on the first layer it's oozing a bit of material and during printing the walls of the next hole wich knocks loose the printed hole causing all kinds of problems. 

it didn't make sence to me for it to be using such a slow combing speed, not sure if it's because of jerk and acceleration either though it shouldn't be.  it's pretty much a short 20mm straight line. 
and was equally supprised that i couldn't find a sepperate setting for the speed at wich it combs
(ps yes i know my real problem is adhesion.  it's a new roll of fillament and it's not propperly dry, yet.

Ghostkeeper, 2021-11-08:
> and the problem is, that it starts on one hole, then combs (since it's not retracting) to the hole right next to it. but it doesn't do it at the travel speed. if i have to guess it's about 45 m/s

On the first layer, it should travel at the Initial Layer Travel Speed. The default for this setting is the normal travel speed adjusted by ratio of Initial Layer Speed / Print Speed. Feel free to adjust the default for yourself. And if you think the default should be improved, feel free to submit a pull request to improve it in the profiles for whatever printer you're using!

> and was equally supprised that i couldn't find a sepperate setting for the speed at wich it combs

Combing is really just travelling while avoiding objects. I don't think it needs a different speed from travels that don't avoid objects. If it needs to slow down when it gets close to objects to avoid, that can be done via acceleration and jerk settings.

darkracer125, 2021-11-08:
I think i explained it propperly in my comment.  Sometimes its combing at a slow speed.  Slower then print speed.    And its deffinatly combing.   So the assumption you made about it combing at travel speed is nothing more then bulls***

Outlook voor Android<https://aka.ms/AAb9ysg> downloaden

Ghostkeeper, 2021-11-15:
Do you have a project file that reproduces the behaviour you describe, then?