Ultimaker/Cura #3050: bugs for Cura3.1

warnant, 2017-12-27:
**Application Version**

**Platform**

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

warnant, 2017-12-27:
- Cura 3.1 :
- Original+

Problèmes :
1-	Nivellement du plateau : la tête d’impression se positionne trop haut par rapport au plateau.
2-	Pas de curseur visible permettant de paramétrer la finesse d’impression (configuration de la couche/hauteur de la couche)
 
Luc Warnant

ChrisTerBeke, 2017-12-29:
Please fill out the issue template and post your comments in English, otherwise we might not take the time to process the issue.

warnant, 2017-12-31:
- Ultimaker Original +
- Cura 3.1

Problems: 
1- Level build plate : the print head is positioned too high in relation to the plate.
2- No visible slider to set the fineness of printing (Print setup / layer height)

Luc Warnant

ChrisTerBeke, 2018-01-01:
The Ultimaker original requires manual levelling of the hardware, guides for this can be found on the Ultimaker website. In Cura you must setup a first printer to see the settings appear in the right sidebar. Could you send a screenshot of your current situation?

warnant, 2018-01-03:
Hello Chris :-)
1- The first adjustments were made manually.
Leveling works in the previous version of Cura. The print head is positioned correctly
[image]
In version 3.1, the print head is too high in relation to the platter.
[image]
2- Original + is selected but the cursor is not present.
In the previous version of Cura, the cursor is present.
[image]

Lu :c)

ChrisTerBeke, 2018-01-03:
For 1 I'll have to ask some colleagues. It might be that the wrong settings are used somehow?
Nr 2 is actually expected because you have changes to the preset profile that belongs to one of these slider ticks. You can see that by hovering over the reset icon (back round arrow) in front of the slider. You can also click that reset button to discard your changes and use one of the preset profiles that belong to the slider ticks.

Ghostkeeper, 2018-01-22:
The bed levelling machine action hasn't been modified for over a year. It depends only on the machine width and height also, so I wouldn't expect that to be a problem. I'll log a ticket to get it figured out.

closed 2018-08-08 by ianpaschal