Ultimaker/Cura #3051: 3.1 Clean Install - How to completely remove Any Cura versions from Win10/64

Thisismydigitalself, 2017-12-27:
I've reached a point where I need to completely clean my PC from any cura installation ever installed so i can at least run 3.1 version solo. Since i upgraded my 3.0.4 to 3.1 Beta i am wasting more time fiddling with it than working with it and I can't install 3.1 release anymore. for some reason it wont run. i was told this is due to the fact that i installed the Beta version which messed up something.

Please direct me what to backup and how to completely remove any(!) version ever installed, including files, folders, registry, instances, etc.. 

I want to start off with a clean slate.

These days I can only run 3.1Beta. 
3.1 release and 3.0.4 will not run anymore.

I don't know when 3.2 is due. if it was few days away i would have waited but even if it will be out as soon as tomorrow i don't want to upgrade from my currently messed up environment.

Sorry. I just want to go back and enjoy Cura like I used to.
Thank you!

fieldOfView, 2017-12-27:
Modern Cura has files in:
* %PROGRAMFILES%\Cura 2.*
* %PROGRAMFILES%\Ultimaker Cura 3.*
* %LOCALAPPDATA%\cura
* %APPDATA%\cura

Thisismydigitalself, 2017-12-27:
Thank you but from experience in my case it was not enough to only uninstall and remove leftover folders. i tried that and still couldn't install 3.1. i guess more is needed. what should i remove from registry?

fieldOfView, 2017-12-27:
Cura does not touch the registry

Thisismydigitalself, 2017-12-28:
**Cura 3.1 release is now running on my PC 👍** 
These are good news. uninstalling all previous Cura versions and deleting the above mentioned folders did  the trick. Now i need to either recreate my profiles and printers and hopefully I am back to square one. I wonder if i need to copy backed up files or to spend the time and created these again from scratch for not bringing in troubles.

What do you say?
Thank you.

closed 2018-01-16 by Ghostkeeper