Ultimaker/Cura #3053: Image help Tips for Newbie ?!

fred2088, 2017-12-28:
is it possible to help image help tips for different option ?

like infill - show 5 different picture with 20%, 40%, 60%, 80%, 100% infile picture ?

Regards

jackha, 2018-01-02:
Do you mean for Layer Height? We didn't think it was very helpful.
Or do you mean a different setting?

thetobby, 2018-01-04:
@jackha  I think he means for a visual aid/image for the infill slider in the easy mode panel. Like having images showing what 50% infill looks like. I think % with a slider like today is enough. Users should look at the model with the 3D viewer to see the infill directly and change the values for true representation of infill on the model and adjust accordingly. Simplifying infill further would remove nessesary basic skills. To play of fred2088 suggetion, perhaps adding a menu for selecting infill pattern in the "easy mode" would be something to consider.

fred2088, 2018-01-05:
just checked that infill have a small png to show the % at the end of selection from cura 3.1
it's very good for novoice user to understand visually,
what if extend this features to other option/selection ?

TIA
Regards

filipgoc, 2018-01-07:
@fred2088 Let's entertain the notion. 

What specific functions/ areas we add a little illustration graphic like it is now for infill? 

- support: culd have a little graphic showing support/ no support
- adhesion: not sure. but could have a little graphic showing the differences between brim/ skirt/ raft (depending on what's on in the custom settings)
- layers: seems a bit self-evident, but that could be a lack of novice empathy. a simple thin vs thick graphic would still add another way of quickly understanding what's happening, and I'm not sure there's any clear downsides. 

so... I guess I'm on boards. it could be a nice touch, and I don't see clear downsides if it's done cleanly without cluttering the ui.

Ghostkeeper, 2018-01-23:
Adding some pictures to the setting descriptions has been a desire we've had on the planning for a very long time. Until it got pushed off because it's older than 12 weeks. I'll file a new ticket then.

Ghostkeeper, 2018-03-21:
And the new ticket got pushed off too.

Ghostkeeper, 2021-05-06:
The Settings Guide plug-in implements this by default nowadays.