Ultimaker/Cura #3054: cura3.1 cpu occupy high in Ubuntu16.04 above.

ghostry, 2017-12-28:
The APPimage file downloaded from the Ultimaker website.
When printing through usb, cpu occupy particularly high.
The same machine, below normal cpu win7.

jackha, 2018-01-02:
How much is "particularly high" in comparison with "normal" ?
The code that is run is the same, possibly it's an OS issue?

ghostry, 2018-01-02:
[image]

ghostry, 2018-01-02:
When using Windows, all core usage rates fluctuate around 10%, not over 50%.

ghostry, 2018-01-02:
Ubuntu, do not open cura, cpu usage at 5%, not more than 20%.Smooth operation, open cura started printing, you can clearly find the system becomes slow.

ghostry, 2018-01-08:
3.0.3 version of the cpu to normal.

closed 2018-03-05 by ianpaschal