Ultimaker/Cura #3055: Feature Request: Conditional Bed leveling

fiveseven808, 2017-12-28:
I was wondering if this was possible, but I would like to have some sort of conditional bed leveling implemented. 

What I mean buy this is for anything 3cm in diameter and smaller, I do not need my entire bed mesh leveled via a G29 command. Just having a G28 command get the center position of my bed is fine. 

I often go in and tweak the G-code manually to omit the G29 command if there's a small object I want to print, but since my code gets sent straight to octoprint, it becomes a little more difficult to tweak it once it's sent. 

The whole idea behind this is saving time. I have a 5x5 mesh bed leveling (my bed is really screwed up), and tweaking g-code on small objects that I only print once can be a waste of time (although a time saver on objects i want to print on a reoccurring basis). 

I would be interested in a feature, specific to printers with auto bed leveling that would be able to recognize the bed footprint of an object, and if it is smaller than a certain customizable size, omit a G29 command and just use the standard G28 command only. Since cura seems to have many experimental features for slicing, I figured it wouldn't be too unreasonable to put a feature request here. 

What do you think? 
Thanks for reading!

ChrisTerBeke, 2017-12-29:
Which printer do you use? The UM3 only levels when needed by default for example, but many other printers just level every time before printing. One way to solve this would be to change the start G-code to use the simpler G28 and manually perform a full mesh levelling when you feel it's needed. The Cura core team primarily builds features for Ultimaker machines, 3rd party machines that require special features have lower priority. Anyone could however make a pull request contribution to implement this.

filipgoc, 2018-01-04:
@fiveseven808
basically, what you're asking for is some sort of toggle to add/remove G29 from the start gcode? 

indeed, you can tweak the start gcode in the Machine Settings from within Cura. It's a bit of a hassle, but if you keep G28 as default and only add G29 when needed, then you should be fine? It would work with the octoprint plugin then.

Ghostkeeper, 2018-01-22:
This goes for Ultimaker printers just as well as for others, even if the Ultimaker 3 has a separate entry in the menu that sends the G29 command.

Allowing a profile to change the start g-code based on the current print's volume is complex. It would make settings dependent on the current print volume. And when they are, settings that influence the disallowed areas could depend on the current print volume as well. And when they are they could cause objects to enter the disallowed areas, which influences the current print volume again. So there is a circular dependency. That might not be a realistic case, but in the order in which things will need to be updated there will be a circular dependency too then, so either something doesn't get updated properly or it ends up in an infinite loop.

filipgoc, 2018-01-22:
Tell us more about your system. Are you running Marlin? 
I'm just thinking now, trying to implement conditional leveling is ... a poor solution. 

It doesn't sound like you have swappable bed. 

So, if you want to save time, then instead generate a good mesh, save it in EEPROM, and then use that mesh instead of generating a new one each time. 

If you're on Marlin, then enable/ setup bilinear or UBL auto bed leveling. Both have the option to save the mesh and then load it instead of re-generating each time. 

Read up more on http://marlinfw.org/docs/features/auto_bed_leveling.html and http://marlinfw.org/docs/features/unified_bed_leveling.html

filipgoc, 2018-01-22:
PS: If you use the bilinear, then it's basically the same as you're used to (UBL is more complicated). 

G29; generate mesh
M500; save the mesh to eeprom - this is the new step

and then instead of G29, use M420 S1 to activate the mesh - in your Cura start gcode. 

Tadaa.

filipgoc, 2018-01-22:
(All of this assumes you're running a reasonably recent version of Marlin)

Ghostkeeper, 2018-02-08:
All of Ultimaker's printers are running Marlin, but it's a version of Marlin that is split off a few years ago, in June 2013.

GregValiant, 2023-09-14:
I'm going back through some older requests.
I think this one could be done in post process.  Cura notes the size of the footprint in the beginning of the gcode.  If the XY area of the footprint were to fall below a user defined minimum then the G29 could be commented out of the StartUp.
This request is 5 years old.  Would anyone actually still need this?

GregValiant, 2024-11-24:
This request is several years old.  I have the post-processor to accomplish what this request is asking for and will post it here if asked.
I'll go ahead and close this feature request.

closed 2024-11-24 by GregValiant