Ultimaker/Cura #3061: Cura 3.1 Wrong Temperature

cwhinton, 2017-12-29:
**Application Version**
3.1.0

**Platform**
MacOS High Sierra 10.13.2

**Qt**

**PyQt**

**Display Driver**
NVIDIA GeForce GTX 780M 4GB- whatever Apple provides

**Steps to Reproduce**

1. Create Custom Printer with the start GCode below. 

   `
M140 S{material_bed_temperature}; Start heating up the base
`
`
G28 ; Home to top 3 endstops
`
`
M109 S{material_print_temperature}
`
   Also set the flavor to RepRap
2.  Activate Custom Printer
3.  Create Custom Material
   Set the print temperature of the custom material to a value other than 210.
4.  Activate the Custom Material
5.  Drop random STL file onto the plate
6.  Observe that the Print Temperature in the side bar agrees with the value given in 3 above.
7.  Save the sliced file to a known location
8.  Open the gcode file in your favorite text editor and observe that the temperature placed on the M109 gcode is 210.

**Actual Results**
see above.  the M109 gcode has a temperature of 210

**Expected results**
The M109 gcode should have the temperature that is shown in the UI as the print temperature which defaults to the temperature given to the material.

**Additional Information**
I don't recall this being an issue with the 3.0 version.

ray-johnsoncode, 2017-12-30:
I'm also using **Cura 3.1** (Dec 5 final buid - **not the beta**) on a Mac (10.11.6), and I'm having the same issue with 3.1.

No matter what the temp is set to, it always comes out as **210**, i've resorted to editing the gcode manually to fix the issue.

3.04 did not have this issue, it's new since 3.1.

But strangely it just started happening. I updated to 3.1 on Dec 15, and the issue did not show up until Dec 24, after I did a system security update (2017-005). Could this update have caused this issue?

If so, why and is there a fix or work-around?

CraigMason, 2017-12-30:
Same issue on Windows, 3.1.0.

Regardless of what material or temperatures I set, the following:

`M109 S{material_print_temperature} ;Nozzle temp`

Always becomes

`M109 S210 ;Nozzle temp`

fieldOfView, 2017-12-30:
Duplicate of #3032, #1296 and others

CraigMason, 2017-12-30:
I saw multiple other issues referring to dual extruders. I'm not using dual extruders, only 1.

fieldOfView, 2017-12-30:
#3032 (and other similar issues) is about single extrusion. All of them were marked as duplicate of #1296, because it is the same underlying problem.

Unfortunately this issue has become a lot more problematic due to an architecture change that made all single extruder printers behave the same way as multi-extruder printers, and as a result all single extruder printers now inherited this multi-extruder issue.

ChrisTerBeke, 2017-12-30:
Indeed a duplicate. The problem got worse in 3.1 as also single extrusion machines now have the issue (hence we prioritized it and fixed it for 3.2).

closed 2017-12-30 by ChrisTerBeke