Ultimaker/Cura #3062: bug - Cura3.1 "sub settings" overriding "main settings" does not work anymore

UltiArjan, 2017-12-30:
cura 3.1.0
win 10

[CableHook.curaproject.3mf.zip](https://github.com/Ultimaker/Cura/files/1594059/CableHook.curaproject.3mf.zip)

When I changed "wall line count" from 3 to 4 cura starts slicing, and then sets it back to 3.
I can change it when setting "wall thickness"

The issue is cura will not let a "sub setting" override a "main setting" anymore. I can set the wall line to 4 only by changing the "wall thickness"

The same issue also occurs with other settings , I can't change top layers (but can change layer thickness) and I can't change "infill speed", but can change it when setting "print speed"

I enclosed the project file.

ChrisTerBeke, 2017-12-30:
Did you use the 3.1 beta before using the stable version?

UltiArjan, 2017-12-30:
yes I did

fieldOfView, 2017-12-30:
The beta left your configuration in a broken state. To repair it, go to Help -> Show configuration folder and close Cura. In the folder that opens, go up one level and remove the folder named "3.1" (and only that folder). When you launch Cura again, your configuration from 3.0 will be re-imported and upgraded.

UltiArjan, 2017-12-30:
ah ok, thanks

fieldOfView, 2017-12-30:
(it is in the release notes)

closed 2018-01-01 by ChrisTerBeke