Ultimaker/Cura #3064: Exturder definition: metadata.position vs extruder_nr

stuartpb, 2017-12-30:
What's the purpose of the "position" field under metadata, versus the extruder_nr under settings / overrides? What would make sense would be for `extruder_nr` to refer to the tool number / extruder train, and "position" to refer only to the extruder's position in the frontend UI order, but it seems like the backend is using them indiscriminately: I tried setting my 0 extruder (with `extruder_nr` = 0) to `position: 1` and vice versa (as my printer numbers its left extruder as 1 and its right extruder as 0, and this would make left and right match in the UI); however, when I tried to print using only the right extruder, the left extruder had the material settings of the right extruder (ie. it was heated), and the material for platform adhesion came out of the left extruder, but the material for the print itself attempted to extrude out of the *right* extruder (which was heated to the left extruder's material settings, so this caused some trouble).

ChrisTerBeke, 2017-12-30:
The position field is used in Cura for assigning settings to the correct extruder. I'm actually not familiar with extruder_nr being used in the internals of the software (@Ghostkeeper ?).

jackha, 2018-01-02:
extruder_nr is used in many settings and it points to this metadata.position. So you could say they are compatible.
@stuartpb if you print with the right extruder only, can you check in the layer view or so that it actually only uses the right extruder? Possibly the brim or only the outer line of the brim is still printed by the first extruder by default.

closed 2018-03-27 by ianpaschal