Ultimaker/Cura #3065: Feature request,  Ironing on every or/and N layers

megablue, 2017-12-30:
I would like to request Ironing not being exclusive to top-most layer but ob every layer or selected interval.
The idea being, ironing fills the microvoids between lines, ironing on every layer will make the parts super strong since the microvoids between extrusion lines are basically eliminated/reduced and bond together better.

ChrisTerBeke, 2017-12-30:
Thanks for re-opening here :)

JonasLehmke, 2018-01-01:
I would love to have this feature to do more transparent prints!

mranvick, 2021-02-23:
This feature would also make sense to print gas- (or at least water-) tight containers, and I would love to see it implemented as well ! Any update on this?

Ghostkeeper, 2021-03-03:
Currently this is, I think, a case where the user thinks they want this but actually doesn't. I've tried printing with ironing every layer (by disabling a check in CuraEngine) and the result is terrible, producing weaker prints with more air gaps than when not ironing, due to massive overextrusion. That's why this is not implemented at the moment. It'd need more research.

mranvick, 2021-03-03:
@Ghostkeeper, indeed, the extrusion setting should probbaly be tuned and the fan setting as well. I recently stumble onto this paper : https://www.mdpi.com/2073-4360/11/9/1487/htm and as you can see on Figure 11 for instance there is some significant air gaps inside the structure. (You may also have a look at [this](https://iopscience.iop.org/article/10.1088/1757-899X/903/1/012008/pdf) )

I guess the idea of ironing at each layer **may** be a solution to this porous structure, and may as well reduce anisotropy (and improve transparency in case of transparent prints). But this might not be **the** solution to simply iron at all layers indeed.

By the way, if you have a blog / website where you posted these tries of yours, or some experimentation results along with the parameters that you used, I would be more than happy to have a look at it!

nubnubbud, 2021-10-30:
@mranvick 

well, it's still all experimental at this point in time. as it stands, even testing if something would help requires a significant amount of work for programming and refining, and as is the case with things like adhesion, there are probably multiple ways to produce satisfying results.

When it comes to airtightness, you can get what you want from significantly thick walls. if you're willing to put gaps inside the walls, and include surface ribbing of the right shape- it should work fine, unless you want zero leakage. in that case good luck with most any plastic polymer.

I'm out of a microscope, but those images I took were simply through an old lens-in-tube microscope from the 60's or earlier. You just shine a light from the bottom, and focus just beneath the surface. if the specimen is clear enough to warrant study, it shouldn't matter how thick it is~... and just to note, those image rulers are in micrometers-crystallization of the brightest looking ones are almost 100% and visible porosity is at almost 0 at that scale.
[image]
in this image, the small imperfections are little gaps left over from sanding. (just below the 40 mark is a gash from a large grain from a 100 grit piece of sandpaper). While there is an occasional bubble or spot of dust, as far as I could tell, it's completely solid to gasses, unless they can invade the crystalline material itself, of course. I feel though we are looking for different things, we can both achieve them through more crystalline prints. clarity = purity and lack of gaps, therefore in this case, clarity can be a test for airtightness as well... supposing the part is thick enough to actually have the right qualities and structural needs as a pressure vessel. (I have luck filling pressurized enclosures with balloons. not only does the balloon keep it airtight, but make dangerous catastrophic failure via explosion far less likely!)

> * a slight expansion / compression of the layer patterns from one layer to the next, in a fashion similar to what is already done by Cura with the "Alternate Extra Wall" to catch the infill with an extra wall. But in your suggestion it would be one layer that is "caught" by the next one by positioning the layers [in quincunx](https://upload.wikimedia.org/wikipedia/commons/thumb/3/36/Optimal_orchard_-_rectangle_quincunx.svg/200px-Optimal_orchard_-_rectangle_quincunx.svg.png).
yes! this is exactly what I mean, but note that it's liquid and lightly pressurized the very moment it's laid down...
[image]
so this would be more accurate. There will still be gaps, though, so it's not a perfect solution. 

> * a concentric layer pattern
actually, if you look closely at my tests, you can see the best results were from rectilinear with no perimeters! no clue why, but from what I can tell,
[image]
it might be that the open ends of the lines allow the air to be pushed out by a piece of plastic, and be replaced like in the drawing above, basically filling it in... but there's also no reason that wouldn't happen with concentric I can tell, so my only theory is shaky at best.

 we have to remember though that any pressure in the system is negligible. it only represents flow and force, but the actual material is still not in a compressible state, disallowing such physics for the most part. PETG is sticky

Ghostkeeper, 2021-11-05:
> edit turned post to make sure you see it: I actually have had some luck with printing clear stuff without ironing!

The best results I've had with transparent prints have always been without ironing. Key is to get the material to melt properly into all the crevices, not having any flow changes (e.g. by having both walls and skin) and sanding/polishing well. Ironing adds new air pockets when it puts another "layer" on top (even if that layer is just dragged material from previously printed lines)
[image]

> Notice how in those tests, all the layers are ALWAYS laid, top-to-top. what if there was a protocol to lay down lines between the ones from the previous layer? I use concentric in all the tests that work, so a shrinking model from layer to layer would also give such results. They would naturally fill in the gaps, and with a wide nozzle and small layer size, we could likely see any gaps disappear completely, as cooling air can shrink them if pockets exist by pulling the plastic in as it shrinks.

Staggering linear patterns improves strength, too, but it's better to keep the focus in this thread to ironing multiple layers instead of just the top one. I don't think this has anything to do with Arachne either.

nubnubbud, 2021-11-30:
> Staggering linear patterns improves strength, too, but it's better to keep the focus in this thread to ironing multiple layers instead of just the top one. I don't think this has anything to do with Arachne either.

I agree, but do note that clear prints both allow you to see gaps and have no internal porosity, which is central to the theme here, as a clear print MUST ALSO "fill the micro-voids between lines", which was his primary interest. Maybe we should simply start a thread on clear/solid/nonporous printing research, 

**back on subject-!**
The only way I could imagine doing this correctly, now, would be a combination of quincunx layering and slight overextrusion with a particular nozzle geometry:
[image]
I have experience accidentally raising the bed, and causing ridges to form from one line getting mashed against the side of the previous. If the nozzle had a thin tip and allowed this more readily, then ironed to cause it, you could theoretically put ridges for the next layer lines to rest in, eliminating the gaps between lines. but even then, it's not too different from normal overextrusion:
[image]
it could feasibly done without ironing The only thing I have been able to think of, that ironing could do but other passes could not, is annealing/stress relief.

fvrmr, 2022-02-01:
We are constantly working on making Ultimaker Cura better. Our community is a big part of that by requesting new features and reporting bugs. 

To be able to focus on the most requested and needed features and bug fixes from our community and from Ultimaker, we have decided to remove this bug from our backlog. We believe we will not work on this anytime soon and will therefore defer this issue.

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.