Ultimaker/Cura #3066: Start Gcode issue

Hebign, 2017-12-30:
I'm using Cura 3.1 and i configured a start script to purge the nozzle before printing:

G28 ; Home all axes
G1 Y10 F3000 ; move Y to 10
G1 Z0.5 F3000 ; lift
G92 E0 ; Reset extrusion
G1 Z0.4 F3000 ; move Z
G1 X100 E30 F600; purge nozzle
G1 Z0.2 F73000; move Z
G1 X150 F5000; wipe

Whenever I start printing and the script goes into operation, nothing comes out during purging, then the print head goes to the center of the tray and starts printing but nothing comes out of the nozzle for about 40s .

Also using Simplify3D, with the same script, I have no problem.

Do you have a solution?

My printer is a Creality CR-10

fieldOfView, 2017-12-30:
Sounds like your printer is not done heating before it starts printing. The firmware prevents extruding before the hotend has reached a certain temperature.

Is this the full start gcode? I don't see anything about heating up the hotend. If you have hotend heatup gcode in there, use M109 instead of M104.

LOVE55, 2018-01-19:
Check whether the extruder is loose, if it is not loosened to give you a suggestion Code:G90
M82
M107
G28 X0 Y0
G28 Z0
G92 E0
G1 X0 Y-100
G1 Z0.2
G1 Y-80 E25 F200
G92 E0
G1 Y-55 E0.1 F100
G1 X11 F100
G92 E0
G1 F1000
My machine is three axis. It's not the same as yours, only for reference.------------------------------------------------------I'm from China, I can't speak English, except the code is Machine Translation.

Hebign, 2018-01-19:
Thanks for the answer, but, no problem with my extruder...

ChrisTerBeke, 2018-05-23:
No new info, seems like incorrect custom settings. Closing.

closed 2018-05-23 by ChrisTerBeke