Ultimaker/Cura #3067: Uncought exception after updating from 2.6 to newest version on ubuntu 16.4 LTS

Elemino, 2017-12-31:
Hello, 

I ran 'sudo apt install cura' to update my cura version. Tried running 'cura' on the command line and got the following exception:

cura.CrashHandler.show [39]: An uncaught exception has occurred!
cura.CrashHandler.show [42]: Traceback (most recent call last):
cura.CrashHandler.show [42]:   File "/usr/bin/cura", line 54, in <module>
cura.CrashHandler.show [42]:     import cura.CuraApplication
cura.CrashHandler.show [42]:   File "/usr/lib/python3/dist-packages/cura/CuraApplication.py", line 59, in <module>
cura.CrashHandler.show [42]:     from . import CuraActions
cura.CrashHandler.show [42]:   File "/usr/lib/python3/dist-packages/cura/CuraActions.py", line 19, in <module>
cura.CrashHandler.show [42]:     from cura.Settings.SetObjectExtruderOperation import SetObjectExtruderOperation
cura.CrashHandler.show [42]:   File "/usr/lib/python3/dist-packages/cura/Settings/SetObjectExtruderOperation.py", line 7, in <module>
cura.CrashHandler.show [42]:     from cura.Settings.SettingOverrideDecorator import SettingOverrideDecorator
cura.CrashHandler.show [42]:   File "/usr/lib/python3/dist-packages/cura/Settings/SettingOverrideDecorator.py", line 14, in <module>
cura.CrashHandler.show [42]:     from cura.Settings.PerObjectContainerStack import PerObjectContainerStack
cura.CrashHandler.show [42]:   File "/usr/lib/python3/dist-packages/cura/Settings/PerObjectContainerStack.py", line 5, in <module>
cura.CrashHandler.show [42]:     from UM.Settings.Interfaces import PropertyEvaluationContext
cura.CrashHandler.show [42]: ImportError: cannot import name 'PropertyEvaluationContext'

Please help! Thanks you so much! 

Elem

Elemino, 2018-01-01:
Maybe I'm missing some python dependencies but now file structure looks different is it not? Anyways, I'm looking into this but if you a quick trick I would love to know!

closed 2018-01-01 by Elemino

fieldOfView, 2018-01-01:
Sounds like you have an outdated Uranium. Cura and Uranium should be updated at the same time.

Elemino, 2018-01-01:
Thanks for the reply! I cloned the uranium repo in my home folder. Then I ran 

export PYTHONPATH=$PYTHONPATH:/home/Uranium/run_mypy.py

Now I get the following error:

cura.CrashHandler.show [39]: An uncaught exception has occurred!
cura.CrashHandler.show [42]: Traceback (most recent call last):
cura.CrashHandler.show [42]:   File "/usr/bin/cura", line 72, in <module>
cura.CrashHandler.show [42]:     app = cura.CuraApplication.CuraApplication.getInstance()
cura.CrashHandler.show [42]:   File "/home/user/Uranium/UM/Application.py", line 338, in getInstance
cura.CrashHandler.show [42]:     Application._instance = cls(**kwargs)
cura.CrashHandler.show [42]:   File "/usr/lib/python3/dist-packages/cura/CuraApplication.py", line 173, in __init__
cura.CrashHandler.show [42]:     ContainerRegistry.getInstance().addResourceType(self.ResourceTypes.QualityInstanceContainer)
cura.CrashHandler.show [42]: TypeError: addResourceType() missing 1 required positional argument: 'container_type'

Am I heading towards the right direction? Will update if I figure out something new. 

Thank you.

fieldOfView, 2018-01-01:
Have you considered using the official AppImage instead? That is the only supported way to run Cura on Linux.

Elemino, 2018-01-02:
It worked thanks! :)

Elemino, 2018-01-02:
Omg! The printer control is amazing!

Ghostkeeper, 2018-01-17:
> it is imperative to type the command with 'sudo' to avoid the program from hanging

That shouldn't be necessary. Cura doesn't do anything that requires super user rights in Linux. Sounds like a bug to me, but it's not reproducing on this end.

Elemino, 2018-01-17:
Let me know if you need more information about my machine for your research. I'd be happy to help and let you know if I figure out the why it does that. I am very happy with the new version. Cheers!