Ultimaker/Cura #3069: Debug/splash image mission when cura crashes

thetobby, 2018-01-01:
**Application Version**
Windows 3.1

**Platform**
Windows, might affect all platforms

**Qt**
Ultimakers download build

**PyQt**
Ultimakers download build

**Display Driver**
Nvidia, latest

**Steps to Reproduce**
Invoke a crash of the application. (can be done with changing quality files from = True , to =Enable )

**Actual Results**
Popup of a blank window with no content.

**Expected results**
 looking at the source I belive it should show a kitten image and some context.

**Additional Information**
I suspect the image is also not compliant with the license Cura is distributed under, at least I find no reference in the commit, perhaps change it to be safe.

nallath, 2018-01-03:
We had to remove the kitten because it was deemed unprofessional. As far as I know, we did use an image that we were allowed to use.

fieldOfView, 2018-01-03:
I don't think OP is complaining about not getting a kitten, but about getting an empty crash handler window (ie: no text either).

nallath, 2018-01-03:
Yeah I know, but that was the thing I could give some info on.

Ghostkeeper, 2018-01-17:
This was a bug with 3.1 that is now fixed for 3.2.

You still don't get kittens, and no professional kittens with little ties either. Nobody to chase the bugs and eat them :cry: But at least you get a crash reporter now with a lovely button to immediately send the crash report to us.

closed 2018-01-17 by Ghostkeeper