Ultimaker/Cura #3070: Controlling the sequence in "One at a Time" printing

brolifen, 2018-01-01:
This is more of a feature request. When choosing to print one at a time it's nice to have more control on the printing sequence by adding a sequence ID perhaps per model which defines the order objects are printed to allow for more user control.

**Application Version**
3.1.0

**Platform**
Windows 10

**Steps to Reproduce**
**Actual Results**
**Expected results**
**Additional Information**

Ghostkeeper, 2018-01-17:
The order in which the objects are printed affects their collision area. We currently edit the collision area to make it as small as possible for which we can still compute an order that prints all of them without colliding the print head to an object.

ianpaschal, 2018-03-16:
Based on the above, I'm going to close this for now. It's a big change to something that isn't really a problem.

closed 2018-03-16 by ianpaschal

redxeth, 2019-02-08:
I think it'd be useful.  Trying to design a multiprint using 20mm squares for bed levelling and would like to control the order that each square prints.