Ultimaker/Cura #3071: [BR]

Witgang, 2018-01-01:
**Application Version**
3.1.0

**Platform**
Windows 10

**Steps to Reproduce**
just throw 2 small objects in and move them away from each other then select "One at a time" and  let it slice then save file.

**Actual Results**
when printing the print head pauses on the last layer of the first printed part then it raises and moves on to print the next part.
  the problem with this is that it leaves a small blob or sometimes even melting the part ruining the finish.

**Expected results**
the print head should raise before waiting for heated bed to reach temp

**Additional Information**
so the  "M190 S60" should be generated after "G0 F7200 X36.633 Y90.646 Z19.35"

Witgang, 2018-01-01:
also(unrelate to the issue) please for god sake increase the auto slicing time, i'm to used to it to turn it off but it gets me every single time i have 2 or more objects can't move one and then the other, i have to wait for slicing between each item move and with 10 or so it's reaaaly painful and is a chore to go disable it then remember to enable it..

ChrisTerBeke, 2018-01-01:
For the 2nd issue (which is gonna be hard to fix as it's related to how the entire slicing engine works), please file a new issue if you want. Keeping track of multiple issues in 1 thread is hard.

Ghostkeeper, 2018-01-17:
I thought we fixed this here: https://github.com/Ultimaker/CuraEngine/commit/f32d2e3972cf9a17f921367a81868c605b22176a

But apparently not. I still see the following g-code with two small objects:

    ;TIME_ELAPSED:620.565057
    G1 F1500 E104.87266
    M107
    M190 S50 ; <- Waiting on temperature
    G0 F7200 X46.466 Y144.169 Z15 ; <- Move up a bit.

Ghostkeeper, 2018-01-17:
It appears that heating up for the first layer temperature works properly, but not cooling down!

Ghostkeeper, 2018-03-21:
Our project manager removed this from our planning. We'll keep it open bug we won't be getting around to debugging this soon.

timangus, 2018-04-18:
That's... an interesting choice. If you're printing something intricate this can ruin the part, right at the last second!

Nuttavoot, 2018-04-19:
This bug still exist.

closed 2023-06-24 by github-actions[bot]