Ultimaker/Cura #3072: exo-open: symbol lookup error: /usr/lib/x86_64-linux-gnu/libpangoft2-1.0.so.0: undefined symbol: FcWeightToOpenType

Lalarian, 2018-01-01:
**Application Version**

3.1.0

**Platform**

Xubuntu 17.10

**Qt**

4.8.7

**PyQt**

4.11.4

**Display Driver**

LibGL Vendor: NVIDIA Corporation
Renderer: Quadro K4000M/PCIe/SSE2
Version: 4.4.0 NVIDIA 340.104

**Steps to Reproduce**

Open any Item in 'Help' menu besides 'About'. Also appears on 'Open Folder' after saving a file.

**Actual Results**

Nothing happens at all.

**Expected results**

A window of the file manager should open, or a website, depending on the chosen item.

**Additional Information**

-

jackha, 2018-01-02:
What do the logs say? It looks like this libpangoft2 is somehow incompatible? I couldn't reproduce it on my Ubuntu 16.04

awhiemstra, 2018-01-02:
I have seen this too on my machine. It is related to the LD_LIBRARY_PATH we set to start Cura from the AppImage. Basically, anything external we launch should not have that set, otherwise it will try to launch the external application with the Cura libraries, which fails.

Lalarian, 2018-01-02:
Which logfile would I have to search for this?

Lalarian, 2018-01-02:
There's no entry when this problem occurs in ~/.local/share/cura/3.1/cura.log

Ghostkeeper, 2018-07-24:
In #3912 it was reported that people get this error when opening any web browser from Cura.

Ghostkeeper, 2018-08-24:
The error for me (on Ubuntu 18.04) only reproduces when running from a build, not from source. That means that we're putting something in that AppImage that has this problem.

Ghostkeeper, 2019-01-22:
The problem seems to be fixed now. I don't know what fixed it though. Recent builds don't reproduce the problem any more.

closed 2019-01-22 by Ghostkeeper

cqwelly, 2019-03-12:
maybe other package had installed pango high version