Ultimaker/Cura #3073: Cura crashes on printer reconnect over octoprint

gitstupid, 2018-01-01:
I've connected Cura over octoprint to my printer. works fine, but if i reconnect printer (in case of: powerfailure or turn off/on printer or reconnect via octoprint)  cura opens a "grey & empty" window and crashes after closing this window

**Application Version**
Cura 3.1 AppImage 
**Platform**
Linux 64bit 4.14.8, XFCE 

**Steps to Reproduce**
connect printer with octoprint
connect cura via api-key with octoprint
disconnect printer over webinterface & reconnect it again
(or turn on/off printer and reconnect...)

**Actual Results**
Cura crashes: Cura spawns a new empty window. you cant use cura because the new window is modal. if you close this empty window - cura also is closed

**Expected results**
cura keeps open? cura save the last state bevor the empty window is displayed?

**Additional Information**
Please open bug reporting for users without GIT-Accounts.  "please register to support..." is just annoying

fieldOfView, 2018-01-01:
Can you post your logs somewhere and add a link to them here?

thetobby, 2018-01-01:
Just a side note, the new empty window you are experiencing is most likely the crash dialog issue mentioned in #3069  The windows is most likely not related to the reason why the application crashed.

Ghostkeeper, 2018-01-17:
3.1 actually had a way to report bugs without Github. Only this new feature of the crash reporting window probably broke the crash reporting window, which is why you're seeing a "new empty window" instead of the normal crash reporting window.

ChrisTerBeke, 2018-01-17:
The original issue described here is about a crash occurring when reconnecting in OctoPrint, not the crash window being non-functional?

ChrisTerBeke, 2018-01-17:
Understandable. Let's see if @gitstupid can reproduce in Cura 3.2 (which will have a working crash dialog again).

gitstupid, 2018-01-20:
Where can I find Cura 3.2?

```
2018-01-20 10:53:03,362 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] [WARNING] There are empty layers (or layers with empty extruder plans) in the print! Temperature control and cross layer travel moves might suffer.
2018-01-20 10:53:04,230 - DEBUG - CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [505]: Slicing took 1.9592790959259953 seconds
2018-01-20 10:53:04,825 - DEBUG - UM.Backend.Backend._logSocketState [182]: Socket state changed to Closing
2018-01-20 10:53:04,829 - DEBUG - UM.Backend.Backend._onSocketError [204]: Socket debug: Arcus Error (13): Closing socket because other side requested close.
2018-01-20 10:53:04,831 - DEBUG - UM.Backend.Backend._logSocketState [184]: Socket state changed to Closed 
2018-01-20 10:53:04,833 - DEBUG - CuraEngineBackend.CuraEngineBackend._onBackendQuit [629]: Backend quit with return code 0. Resetting process and socket.
2018-01-20 10:53:04,924 - DEBUG - CuraEngineBackend.ProcessSlicedLayersJob.run [237]: Processing layers took 0.7905313096079199 seconds
2018-01-20 10:53:06,985 - WARNING - cura.PrinterOutputDevice.address [269]: address is not implemented by this output device.
2018-01-20 10:53:06,987 - WARNING - cura.PrinterOutputDevice.address [269]: address is not implemented by this output device.
2018-01-20 10:53:06,988 - WARNING - cura.PrinterOutputDevice.address [269]: address is not implemented by this output device.
2018-01-20 10:53:06,990 - WARNING - cura.PrinterOutputDevice.address [269]: address is not implemented by this output device.
2018-01-20 10:53:10,233 - DEBUG - AutoSave.AutoSave._onTimeout [48]: Autosaving preferences, instances and profiles
2018-01-20 10:53:48,902 - WARNING - cura.PrinterOutputDevice.address [269]: address is not implemented by this output device.
2018-01-20 10:53:48,904 - WARNING - cura.PrinterOutputDevice.address [269]: address is not implemented by this output device.
2018-01-20 10:53:48,905 - WARNING - cura.PrinterOutputDevice.address [269]: address is not implemented by this output device.
2018-01-20 10:53:48,907 - WARNING - cura.PrinterOutputDevice.address [269]: address is not implemented by this output device.
2018-01-20 10:57:10,376 - DEBUG - SliceInfoPlugin.SliceInfo._onWriteStarted [59]: 'info/send_slice_info' is turned off.
2018-01-20 10:57:10,400 - WARNING - cura.PrinterOutputDevice.address [269]: address is not implemented by this output device.
2018-01-20 10:57:10,401 - WARNING - cura.PrinterOutputDevice.address [269]: address is not implemented by this output device.
2018-01-20 10:57:10,403 - WARNING - cura.PrinterOutputDevice.address [269]: address is not implemented by this output device.
2018-01-20 10:57:10,405 - WARNING - cura.PrinterOutputDevice.address [269]: address is not implemented by this output device.
2018-01-20 10:57:10,806 - DEBUG - OctoPrintPlugin.OctoPrintOutputDevice._onRequestFinished [770]: Resource created on OctoPrint instance: http://octoprint.lan:80/api/files/local/CC0.4_Echo.gcode
2018-01-20 11:00:15,451 - WARNING - cura.PrinterOutputDevice.address [269]: address is not implemented by this output device.
2018-01-20 11:00:15,453 - WARNING - cura.PrinterOutputDevice.address [269]: address is not implemented by this output device.
2018-01-20 11:00:15,455 - WARNING - cura.PrinterOutputDevice.address [269]: address is not implemented by this output device.
2018-01-20 11:00:15,457 - WARNING - cura.PrinterOutputDevice.address [269]: address is not implemented by this output device.
2018-01-20 11:02:35,278 - CRITICAL - cura.CrashHandler.__init__ [62]: An uncaught exception has occurred!
2018-01-20 11:02:35,280 - CRITICAL - cura.CrashHandler.__init__ [65]: TypeError: unable to convert a Python 'NoneType' object to a C++ 'double' instance

```

in cura 3.1 i can still reproduce the error
connect Cura to Octoprint, then connect Octoprint with printer.
If I know disconnect the printer in Octoprint and reconnect the printer (also via Octoprint) => Cura crashes...

with cura 3.0.4 i couldnt reproduce the crash

ChrisTerBeke, 2018-01-21:
3.2 beta is due this week. as it's a beta I recommend keeping 3.1 next to it for any production purposes.

fieldOfView, 2018-06-11:
The OctoPrint plugin has been refactored for Cura 3.3, and this issue does not reproduce for me with Cura 3.3 or Cura 3.4.

ChrisTerBeke, 2018-06-11:
I'll close it until someone reports the problem again for 3.4+. Thanks @fieldOfView for the fixes :)

closed 2018-06-11 by ChrisTerBeke

jan-panoch, 2018-11-21:
hello,

it's here again - cura 3.6, octoprint plugin 3.5.2 and octoprint 1.3.9

2018-11-21 13:58:49,542 - INFO - [MainThread] cura.Machines.QualityManager.getQualityChangesGroups [179]: Cannot find node for machine def [fdmprinter] in QualityChanges lookup table
2018-11-21 13:58:49,544 - DEBUG - [MainThread] cura.Machines.Models.QualityProfilesDropDownMenuModel._update [50]: Updating QualityProfilesDropDownMenuModel.
2018-11-21 13:58:52,899 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [181]: Error check finished, result = False, time = 4.3s
2018-11-21 13:58:58,322 - DEBUG - [MainThread] cura.AutoSave._onTimeout [48]: Autosaving preferences, instances and profiles
2018-11-21 13:59:02,218 - DEBUG - [MainThread] cura.Machines.Models.QualityManagementModel._update [38]: Updating QualityManagementModel.
2018-11-21 13:59:02,220 - INFO - [MainThread] cura.Machines.QualityManager.getQualityChangesGroups [179]: Cannot find node for machine def [fdmprinter] in QualityChanges lookup table
2018-11-21 13:59:02,281 - DEBUG - [MainThread] cura.Machines.Models.QualitySettingsModel._update [73]: Updating QualitySettingsModel.
2018-11-21 13:59:02,284 - DEBUG - [MainThread] cura.Machines.Models.QualitySettingsModel._update [73]: Updating QualitySettingsModel.
2018-11-21 13:59:02,362 - DEBUG - [MainThread] cura.Machines.Models.QualitySettingsModel._update [73]: Updating QualitySettingsModel.
2018-11-21 13:59:15,798 - DEBUG - [MainThread] UM.Controller.setActiveStage [159]: Setting active stage to MonitorStage
2018-11-21 13:59:15,908 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [65]: An uncaught error has occurred!
2018-11-21 13:59:15,910 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]: TypeError: unable to convert a Python 'NoneType' object to a C++ 'double' instance
2018-11-21 13:59:15,920 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [352]: file:///Applications/Ultimaker Cura.app/Contents/MacOS/resources/qml/Menus/ConfigurationMenu/PrintCoreConfiguration.qml:73: TypeError: Cannot read property 'name' of null

fieldOfView, 2018-11-21:
Can you open a new issue at the OctoPrint issue queue?
https://github.com/fieldofview/octoprintplugin/issues

Also note that a new version of the plugin has just been published to the ~Toolbox~Marketplace.