Ultimaker/Cura #3076: Cura 3.1 does not show printer def.json files added to app or user Library.

gregschumacher, 2018-01-01:
[cura.log](https://github.com/Ultimaker/Cura/files/1596317/cura.log)

**Application Version**
3.1.0

**Platform**
macOS 10.13.2

**Qt**
?

**PyQt**
?

**Display Driver**
?

**Steps to Reproduce**
Quit Cura.
Copy Ultimaker\ Cura.app/Contents/Resources/resources/definitions/prusa_i3.def.json to ~/Library/Application\ Support/cura/3.1/definitions/new_printer.def.json.
Edit new_printer.def.json by modifying id, name and overrides.machine_name.default_value.
Launch Cura.
Select Settings > Printers > Add Printer...
Reveal Other printers.

**Actual Results**
New printer does not appear on the list.

**Expected results**
New printer should appear on the list.

**Additional Information**
Also tested saving new_printer.def.json in Ultimaker\ Cura.app/Contents/Resources/resources/definitions/ with the same results.

fieldOfView, 2018-01-01:
Check the logs. There's probably an entry in there that tells you why the definition cannot be loaded.

fieldOfView, 2018-01-01:
The log shows the following error:
UM.Settings.DefinitionContainer.IncorrectDefinitionVersionError: Definition uses version 1 but expected version 2
You are using a json file as your template that is too old.

thetobby, 2018-01-02:
Can you please try to delete/remove roaming/cura and local/cura in %AppData%  ?
 Make backup if you are not willing to lose everything cura related. Since the error claims it to be wrong, just manually veryfy that you do have:

`{
    "version": `2,``

and it is not missing or saying 1

closed 2018-01-17 by ChrisTerBeke