Ultimaker/Cura #3077: Combing Feature

CCS86, 2018-01-01:
Hi guys,

I think it would be valuable to add the following features when combing is enabled:

- Maximum combing move length: combing moves longer than this value switch to retracted, point-point, travel moves

- Combing makeup volume: this value would be made up in an extruder move, at the end of the combing move (similar to retract extra prime). If it could be made a function of combing move distance, that would be even better. Linear would be okay, but I bet it's closer to inverse square.

smartavionics, 2018-01-01:
On a related note, this is what I said the other day when commenting on another issue:

> But still, I can't help wonder whether a sort of (optional) semi-retract could be beneficial during combing. Just a small amount of retract that is carried out concurrently with the initial part of the travel move. At the moment, retract/primes always happen with the nozzle stationary. What I am suggesting is that as soon as the travel move is started at the beginning of the combing sequence, at little bit of retraction could occur to reduce the pressure in the nozzle. At the end of the combing travel, the same amount of filament could be extruded so that the net movement of filament is zero. Would that make combing be usable for runny materials?

> 

>

CCS86, 2018-01-02:
I do like that idea as well. 

Coasting is helpful in bleeding off extruder pressure. But it's not a complete solution. I'm trying to get the newer "linear advance" function working, from Marlin 1.1, but it's a bit hyperactive at the moment.

Ghostkeeper, 2018-01-17:
We've had similar experiments going on to reduce oozing in extremely long travel moves or with large nozzle sizes, but no real success yet as far as I know.

Ghostkeeper, 2018-01-17:
Regarding the maximum combing distance, isn't it better or simpler to understand to add a maximum travel distance to not retract. If there is a move, even a combing move, that is longer than this distance then the move will be retracted (and it will still try to comb).

CCS86, 2018-01-17:
I guess that my thought is, that if it is a retracted move, there is no point in combing. Combing moves are more complex and take longer to complete.

smartavionics, 2018-01-17:
Hi, i put together some experimental code which did a "negative extrude" at the start of the travel move and then a "positive extrude" at the end. This is different from a retract/move/prime because the filament moves for some of the time while the nozzle is moving. I only tested it briefly but found that unless the travel distance was quite large, it wasn't travelling for long enough to allow for a reasonable amount of retraction and it would still ooze at the start of the move. I could do some more work on that but it didn't look very promising.

I think the best thing would be to have max comb distance setting. If a combed travel is >= that distance, then it just does a vanilla retract/move/prime in a direct line between the two points. If the distance is less, then it should comb with no retract.

pynej, 2018-01-31:
For my money the necessity here isn't changes to try and reduce the visibility of combing.  Ie if surface finish is critical then turn off combing. But rather to allow for composition due to material lose on the combing.  When running at slow speeds and thus lower temp this isn't a problem, but when running at faster speeds and thus higher nozzle temp the oozing/lose on the move eventually causes inner wall failures and print failures.  

A Extra Prime on Combing Move is needed.

ianpaschal, 2018-03-27:
Closing because the conversation has been continued in #3245.

closed 2018-03-27 by ianpaschal