Ultimaker/Cura #3078: Infill extruder overriden by top/bottom extruder

MB3Y, 2018-01-02:
**Description/Steps to repro**
I wanted to print a transparent Shell, with darkblue infill such that the infill becomes "visible" looking at the object.
I select Extruder1 for Shell, Extruder2 for infill. Nothing happens! Still Extruder 1 is used for infill material.
But when I change the Top/Bottom Extruder to Extruder2, then the infill material is changing to the top/bottom material (infill extrucer is not overriden when this happens)
Playing with both extruder values has at times resolved (only once in fact) this issue for me, but I can't figure out what is the dependent factor.
I checked that it is not the representation of the material view itself: I can confirm that the Gcode also does not contain the T0/T1 commands, i.e.: they are there, but only for the bi-colored skirt.

**Application Version**
3.1.0

**Platform**
Win10

MB3Y, 2018-01-02:
Got it!
Setting infill density to 100% makes such that the top:bottom material/line type is expanded over the whole object!
When chosing 99% infill (or lower), then the infill is actual infill and all works as I expect.

Is this a bug or is there some reason that 100% infill behaves as top/bottom line types??

PS: thing printing is the labyrith lid of [https://www.thingiverse.com/thing:201097/#files]

smartavionics, 2018-01-02:
> Is this a bug or is there some reason that 100% infill behaves as top/bottom line types??

I don't believe it is a bug but I can't remember the rational behind using skin for 100% infill. A Cura dev will be able to answer that.

jackha, 2018-01-02:
It is not a bug. If you set infill to 100% it will actually increase the bottom layers to 99999 (something). This way the model will print better. You can manually override this.

closed 2018-01-02 by jackha

fieldOfView, 2018-01-02:
> the model will print better this way

By printing everything as bottom, the layer is printed as one contiguous pattern, instead of a contiguous pattern with bottom/top "parts" around it (eg in the case of minor inclines in the walls).

You can override the behavior by specifying the number of bottom layers.

MB3Y, 2018-01-02:
Thank you guys for the explanation, there is indeed a dependency with the bottom layers - and SOME of it is indicated in the help text.
My suggestion is to
- update help text for top and bottom "thickness"
- update fields for top/bottom thickness as well as top/bottom layers when fiddling with infill density
- remove the precedence of layers over thickness (both should be perfect alternatives)
Below how I debugged my problem further and why I suggest the above.

-------
Unfortunately there is still a bug -in my view-: the value for the bottom layers is not updated in the UI to represent this behavior:
1. I have 1mm bottom tickness -or 6layers @0.17mm-
2. then I change infill to 100%
=> bottom layers stay at 6, but infill becomes skin

3. Change Bottom thickness to 0.8mm
=>Bottom layers are again not updated (and then I figured: this is only if bottom layers were set by hand before, when they are on default, then the update is done to 99999!)
Q: why are bottom layers and bottom thickness not coupled always?

4. Change Bottom Layers to 6 (/some other value than it was before)
=> Now it becomes apparent that also the Top Layers are in fact gone - and oops: the top thickness nor layers are updated
=> same behavior as with bottom thickness/layers: only when changing the layers value the gcode generated is different.

Anyway: thanks for the explanation, I know now a workaround :).

Ghostkeeper, 2018-01-18:
We've gotten this sort of question before. It's unclear that it happens ("why doesn't my infill speed get used at 100% infill?!?") and why. So if there is a better way to communicate this in the user interaction then we'd like to hear it.

The tooltips for infill density and bottom layers (automatically) show that the bottom layers is affected by infill density. But I think we could add some clarification to the tooltip of infill density then. Maybe that helps a few of the confused people.

I have long wanted to have some sort of way to select the unit of a setting, instead of the current precedence of layer count over mm. You'd be able to select the unit in which the setting is displayed, either mm or layer count, and under water just the layer count is sent to the engine for slicing. The real problem is that it's difficult to fit more interface elements in the very limited amount of space in the settings list there.

This is also the cause of that bug.