Ultimaker/Cura #3080: Gcode viewer only shows colors for T0 and T1

MB3Y, 2018-01-02:
There should be a color for T2 (and more) also
e.g. T2 for diamond hotend emulation
e.g. T3 for Prusa MMU emulation

ChrisTerBeke, 2018-01-08:
Can you fill out the issue template and post a screenshot?

MB3Y, 2018-01-08:
I'll be glad to add screenshots, but didn't figure out how to do it yet. Should I upload somewhere and link to it?

Sorry not to have used template, will update later this evening. Issue seemed so straight forward to me (add T2 in Gcode and load up to Cura)

MB3Y, 2018-01-08:
**Application Version**
3.1

**Platform**
Win10

**Steps to Reproduce**
File - Open - Something.gcode

**Actual Results**
Representation for T0 and T1 is Ok, T2 and higher are not show / as transparent.

**Expected results**
I expect to see a representation for each extruder.

**After slicing using standard only 1 extruder**
[image]

**After striping in 12 equal parts using 3 extruders**
[image]

ChrisTerBeke, 2018-01-08:
@diegopradogesto any ideas given that you worked on the layer view recently?

diegopradogesto, 2018-01-09:
It seems not to be related with layer view but with the GCode reader.
Devs: CURA-4778

ChrisTerBeke, 2018-01-09:
Fixed in https://github.com/Ultimaker/Cura/commit/e3e6f301f9172c6bf811ac5ee432c277ba08deb2

closed 2018-01-09 by ChrisTerBeke

MB3Y, 2018-02-23:
It definitely is ! 👍 
[image]