Ultimaker/Cura #3083: Ironing

kartorschkaboy, 2018-01-03:
If I check "ironing", curaengine.exe will crash when it tries to finish the slicing, the programm stays open and funktions normaly, but it can't slice with "ironing" checked... but when I delete the cura save file in appdata/local (I guess it was there) and restart Cura and set everything up again it works again, but after a while the same issue occures...

is there a setting that interferes with "ironing"?

LipuFei, 2018-01-03:
@kartorschkaboy There is a known issue regarding this: https://github.com/Ultimaker/Cura/issues/2932
Could you try disabling "Remove Empty First Layers"?

Ghostkeeper, 2018-01-22:
Try this again with the 3.2 beta tomorrow. It should be fixed via https://github.com/Ultimaker/CuraEngine/commit/e79893968d5c6497b7d1c0e1015722b07f38b3b5.

closed 2018-01-22 by Ghostkeeper