Ultimaker/Cura #3084: building without appimage?

clytle374, 2018-01-03:
On Gentoo with sources from git. Is it possible to build without having appimagekit installed? Looks like appimagekit might be a bit of a mess to install. 
Thanks
Cory

jackha, 2018-01-04:
If you only want to run it and not make an AppImage, then you don't need the appimagekit.

clytle374, 2018-01-04:
That is what I am hoping to do.  How is that disabled in cmake to configure without appimagekit?

nallath, 2018-01-08:
It's python. You don't need to build at all (The engine, libArcus & libSavitar need to be build). 

So runinng it can be done with "python3 cura_app.py"

ianpaschal, 2018-02-27:
This question has been answered. [More information can be found in the wiki.](https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source)

closed 2018-02-27 by ianpaschal