Ultimaker/Cura #3085: Cura freezing on setting change

SpaceVoyager30, 2018-01-04:
cura 3.1.0

When every i change a setting in the custom print setup tab or printer settings, 2 secs later cura will freeze for about 10-20 secs. task m10nager shows that cura shoots up to 33.3% cpu usage maxing out one of the four cores. apart from that cura runs smoothly.

system:
Computer: Toshiba Satellite L50D-C
OS: Windows 10 x64
CPU: AMD A4-7210 APU with AMD Radeon R3 Graphics @ 1.8 GHz
GPUA: AMD Radeon R3 1GB @ 0.8 GHz
GPUB: AMD Radeon R5 M330 2GB @ 1.03 GHz
RAM: DDR3 8 GB (1 GB Hardware reserved) @ 1.6 GHz
SSD: 120GB

Driver Packaging Version
17.30.1091-170918a2-318192E-CrimsonReLive

here is the log cura outputs with auto slice off:
steps:
1.open cura
2.change fans settings
3.slice
4.quit
[cura.log](https://github.com/Ultimaker/Cura/files/1602591/cura.log)

alekseisasin, 2018-01-04:
Did you try to install earlier Cura version 2.7 or 3.0.

SpaceVoyager30, 2018-01-04:
Cura 3.0 is the same. Cura 2.7 is a bit better but it still freezes for about 5-10 secs

alekseisasin, 2018-01-04:
I might only guess what problem it can be. Maybe you don't have enough free space on your SSD drive.

SpaceVoyager30, 2018-01-04:
~10 GB free

alekseisasin, 2018-01-05:
I think the problems is related to the video card (maybe conflict between integrated and dedicated video), what exactly hard to say, remotely I cannot test it.

SpaceVoyager30, 2018-01-06:
idk. deleting some stuff helped to the point where it is useable.
i can't do any thing about the video card thing though becauce toshiba thought it was a good idea to put amd switchable graphics on the computer and not add a BIOS option to disable it.

Thank you for the help

mbuc, 2018-01-09:
I'm having the same issue with the latest release.  Every time I change a setting it freezes for about ten seconds.  I have a new DIY printer to configure so this makes it essentially unusable at the moment.  I experience high CPU usage for the app on a ~2014 i7.  It seems to spike in activity as well when tooltips are activated.

I'm also unable to change settings for a custom material under Preferences.  I'll be submitting an issue for that separately.

SpaceVoyager30, 2018-01-09:
Using Cura 2.7 did help a bit for me.

nallath, 2018-01-10:
I did notice something odd; At some points it tries to start the engine a ton of times and repetitively closing it again. It could have something to do with that.

SpaceVoyager30, 2018-01-10:
The Cura engine process that runs in the background?

mbuc, 2018-01-11:
Would a ProcessMonitor log help @nallath ?

fieldOfView, 2018-01-11:
@mbuc, what would be helpful is a cura.log from you, to see if yours shows the same repetitive closing and restarting of the CuraEngine.

SpaceVoyager30, 2018-01-12:
have a look at my log back at the top.
I can see repeated lines of "INFO - UM.Backend.Backend.startEngine [75]: Started engine process: C:\Program Files\Ultimaker Cura 3.1\CuraEngine.exe"

ChrisTerBeke, 2018-01-12:
We're actively looking into and fixing this issue right now :)

mbuc, 2018-01-12:
@fieldOfView I've attached two log files.

File: [cura_18.01.12a.log](https://github.com/Ultimaker/Cura/files/1625826/cura_18.01.12a.log)
Steps to Reproduce:

- Opened Cura
- Waited for Cura to finish loading
- Closed Cura

File: [cura_18.01.12b.log](https://github.com/Ultimaker/Cura/files/1625827/cura_18.01.12b.log)
Steps to Reproduce:

- Opened Cura
- Opened Preferences > Configure Cura...
- Navigated to Printers > Tesla [My custom printer] > Machine Settings
- Changed the X Width from 200 to 201
- Selected the Y Width textbox to trigger setting change
- Experienced freezing for about 4 seconds
- Changed the X Width back to 200
- Selected the Y Width textbox to trigger setting change
- Experienced freezing again for about 4 seconds
- Closed Machine Settings dialog, closed Preferences dialog, closed Cura

fieldOfView, 2018-01-12:
@mbuc thanks, but that's not the issue we are talking about here. The issue is about changing settings in the sidebar. Machine Settings is a substantially different beast.

mbuc, 2018-01-12:
> When every i change a setting in the custom print setup tab or printer settings...

Isn't "printer settings" what I'm referring to?  Not trying to argue, just pointing out that they may be related.

SpaceVoyager30, 2018-01-13:
I was originally referring to both(print setup side bar and the machine settings) but it looks like we are focusing on the print setup side bar at the moment.

jackha, 2018-01-23:
One of the performance issues has to do with error checking all possible extruders for errors. The Custom FDM Printer has 8 extruders and it would check all of them. I fixed that in the upcoming 3.2 release.
Apart from that, the checking itself is also quite slow, but that's more risky to change. However, with the current fix it should be a lot faster already.

fieldOfView, 2018-01-24:
@jackha, it does not seem that validating all extruders is the (only) issue:
https://www.reddit.com/r/3Dprinting/comments/7sevgu/cura_32_beta_available_with_adaptive_layer_height/dt5q7ei/

jackha, 2018-01-24:
@fieldOfView Thanks for mentioning. I've added some logs on those validation checks. If the computer is just a bit too slow it would sometimes validate multiple times which makes the problem even worse :-(

Ghostkeeper, 2018-02-12:
Did it improve with the 3.2 release?

SpaceVoyager30, 2018-02-12:
Cura 3.2 changing print setting is a lot better now, changing printer settings is better but still freezers for 1-4 sec. One thing I noticed is that there are a bit graphicl ui but apparently from that it is a lot better now.
Thank you

ianpaschal, 2018-02-27:
Since there's not a clear point of "Fixed" I'm going to consider "a lot better now" to be fixed.

closed 2018-02-27 by ianpaschal