Ultimaker/Cura #3087: master branch - gcode filename doesn't get updated for subsequent models

smartavionics, 2018-01-04:
**Application Version**

master branch

**Platform**

Linux

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**

Load a model, slice it, note that the gcode file name is based on the model file name. Clear the build plate and load and slice another model.

**Actual Results**

The gcode file name doesn't change (it's still based on the first model's file name not the second).

**Expected results**

The gcode filename should be based on the name of the model just loaded. I appreciate that this becomes more complex when there are multiple models on the build plate but I'm talking about just having a single model at a time.

**Additional Information**

ChrisTerBeke, 2018-01-05:
@jackha due to multi-buildplate merge?

smartavionics, 2018-01-15:
This looks like it has been fixed, thanks. Closing...

closed 2018-01-15 by smartavionics