Ultimaker/Cura #3088: Using and Creating profiles for new non ulitmaker materials not possible

sebther2, 2018-01-05:
Dear all!

By using CURA 3.1.0 (with an ultimaker 3) it is not possible to create / copy profiles (fine / normal etc.) for own created materials (this means non ultimaker materials).

Following steps to reproduce the issue:
- Copy e.g. Ultimaker (also generic) CPE
- Rename the Material to Black PETG
- Company / brand - change from ulitmaker to e.g. renkforce
- Materialtype - PETG

After activating this material it is not possible to use profiles for the printing setup in Cura.
Also it is not possible to create and save a new profile for this non ulitmaker material.

This issue will also happen, if you cancel all relationships to the original (copied) ultimaker material.

[image]

Please contact me for further information!

Helmut

Spirou42, 2018-01-09:
I have the same issue (Cura 3.1, UM3 with 4.2.5.20180102) on MacOS Sierra. It does not matter if the Material was created by the [COPY] or by [CREATE] function. Also it does not matter which material you use as a source for your custom material.

On my Machine the profile, or parts of it, is actually saved into ~/Library/Application Support/cura/3.1/quality but the UI does not allow to load it again. 

Contents of cura/3.1/quality
[image]

UI:
[image]

[image]

Spirou42, 2018-01-10:
Some news on this:
I just created a custom entry for ColorFabbs SteelFill by duplicating "Ultimaker Silver PLA". Changed the name, brand, and material type as well as some of the print parameters while Silver PLA is selected AND the RFID tag of the Silver PLA is ON the printers reader.
In this particular case the Material Dropdown got not coloured orange and the profile stays the same. 
BUT if i try to create a profile from the currently made settings: The dialogBox is empty
 
[image]

and after saving, the profile does not show up in the material dialog

ChrisTerBeke, 2018-01-14:
Devs: CURA-4796
Community: we're picking this up asap as it's a regression bug.

Spirou42, 2018-01-23:
After first quick look into cura3.2b this seams to be fixed. I can see and switch profiles in both, the preferences dialog and the drop down menu. Creating new profiles and switching profiles works as expected i think. 

So the material drop down menu stayed orange and a little exclamation sign told me something about "incompatible materials". All materials I have created where copies from existing generic materials which are still linked to the source material.
[image]
The orange marker occurs for all my materials (all Materials where created with cura3.1)
I have linked
CPE -> nGen
CPE+ -> HT
PLA -> SteelFill (PLAH)

- Carsten

jackha, 2018-01-24:
@Spirou42 Good that the regression is gone. Could you provide a project file so we can produce this orange compatibility issue?

Spirou42, 2018-01-24:
@jackha Absolutely, to both the statement and the question. 

[TestProjects.zip](https://github.com/Ultimaker/Cura/files/1660107/TestProjects.zip)
Contains 3 different projects using 4 different custom Materials.

btw.: the graphics performance (zoom, rotate pan) of Cura3.2b is abysmal compared to  3.1 . 
Should I create another Issue for this or is this due to debug settings in the beta?

screen recording showing performance issue is at: https://youtu.be/05LhF0BR3PY

jackha, 2018-01-25:
@Spirou42 Thanks! You should in this case keep the material type to PLA, it was changed to PLAG and therefore it was looking for profiles for that new type of material.

For the slowness you can file a new bug. I fixed something, but that was only applicable to Custom FDM Printer.

Spirou42, 2018-01-27:
@jackha I have changed the material types back to the original value of the linked material. After reloading of the material, the compatibility warning disappears. 

Unfortunately this has another unwanted UI impact. 
The material type not only is a marker for the compatibility but it is also directly used in the material selection menu - leaving me with a single CPE entry for two different materials (nGen and nGen Flex) and also with a single CPE+ entry for HT and XT. The same problem occurs with FormFuturas Atlas and PVA as support materials.  

Maybe the check for compatibility can also take the ancestors material type into account. Such a behaviour would allow the creation of an arbitrary UI hierarchy by material name and would eliminate the compatibility warning. 

I don't have anything against a safe and robust setup that prevents inexperienced users from doing any harm. But i really would appreciate an "expert mode". Such a mode should - in my opinion - disable all warnings in Cura and also on the UM3 so that i don't have to acknowledge any slight and wanted discrepancy to your very safe setup. The flag could be secured by a method like github uses to prevent deletion of a repository.

jackha, 2018-01-29:
@Spirou42 The selection menu uses the display name, if supplied. You can also change the brand field, so it's in its own sub menu. Only the material type field is used to match profiles. Hope this helps.

An "expert mode" without warnings would be handy indeed.

Spirou42, 2018-01-29:
@jackha Yeah, they are used in the menu hierarchy too. Unfortunately it is a threefold hierarchy consisting of:
BrandName->MaterialType->DisplayName at least on the Mac.

[image]

Due to the current implementation all PLA variation of one Brand has to be sorted into a single MaterialType section including ALL the color variations and material specialities as metal, clay or wood filled materials. 
This leaves the value of MaterialType with two completely different use cases.
1. it marks a material as compatible - machine knowledge 
2. it organises a materials in the UI - UI/UX 

So I have the choice between the devil and the deep blue sea. Either i stay with an incorrect and annoying compatibility marker or i have to use a material hierarchy that is insufficient to keep the zoo of different materials organised.

jackha, 2018-01-30:
@Spirou42 you're right, it's annoying. Currently there's no good solution for this.
You could possibly misuse Brand?

Spirou42, 2018-01-30:
Yeah i think this misuse of the brand name might be a work around for this issue.

Ghostkeeper, 2018-02-21:
It marks the material or nozzle orange when it has no quality profiles for this combination. Because if there are no profiles, Ultimaker can't vet its quality. The printer itself will also give a warning when printing via removable medium.

jackha, 2018-02-22:
@Spirou42 quality profiles refer to the machine / variant / material (i.e. UM3 / AA 0.4 / PLA) specific settings which can be found in resources/quality/<quality profile>. Look at some existing ones, preferably Ultimaker ones (those are the one we create and test thoroughly), and see if you can create your own.

Ghostkeeper, 2018-03-13:
With "quality profiles" we mean the profiles that you select in the drop-down at the word "Profile:". You can select from different qualities there, "Fast", "Normal", "Fine", etc. These profiles are specific to a material, so if you have your custom material then there won't be any quality profiles for it.

The original problem in this issue is not yet solved, even after now having merged our machine manager refactoring.

diegopradogesto, 2018-07-03:
Is this issue fixed? If so, let's close it. Otherwise we need to create a new internal ticket for that.

Ghostkeeper, 2018-08-01:
The original issue is now also fixed. You can create a custom profile for your new material type.

closed 2018-08-01 by Ghostkeeper