Ultimaker/Cura #3090: kozentrische Füllung

Timechild2013, 2018-01-05:
Hallo Community,
Ich habe immer wieder Probleme mit der konzentische Füllung 100% bei runde Teile. An manche Stellen slice Cura egal welche Version lauter kleine Pünkchen/Linie im Zusammenspiel mit Eilgang. Der fährt also mit 60mm/s hin und fährt ein paar Zehntel mit Eilgang 200mm/s zurück. Das ständige hin und her fahren führt natürlich zum Schrittverlust, egal welcher Drucker es auch sein mag, je höher der Eilgang ist desto sicherer Schrittverlust. Ihr könnt einfach irgend welche runde Teile nehmen und ausprobieren.

thetobby, 2018-01-06:
If possible please write in English or create a short English summary. Perhaps you have a friend that can help you translate it. Perhaps someone even speaks german here, but any attempt is appreciated.

Timechild2013, 2018-01-06:
Hello Community,
I always have problems with the 100% infill concentration of round parts. In some layers slice Cura no matter which version small punk/line in conjunction with rapid traverse. So it prints with 60mm/s and drives back a few tenths with rapid traverse 200mm/s. The constant back and forth course leads to loss of steps, no matter which printer it may be, the higher the rapid traverse the safer step loss. You can just take any round parts and try it. Google translate.

thetobby, 2018-01-06:
Under experimental, try enabling "_Optimize wall printing order_".
What printer and settings are you using?
Any pictures?

Timechild2013, 2018-01-06:
I have try any option, but nothing change, only if change wall line to 0.22 or so, but this option make new problems. Infill is contrentration 3D, if only concentration without 3D the infill lines takes outer line speed, this is not normal. Printer is Prusa I3 pro
0.3 nozzle, 0.2 layer, tries 1-200 wall/infill lines.

[https://drive.google.com/open?id=11pRROm59CsRcoY3dYNUoxBmuEAcH6jRm](url)
[https://drive.google.com/open?id=1CFwSNaT-HmlhmeBtKNTilbiQKvlcxL9_](url)

MB3Y, 2018-01-14:
Skipped steps on very fast moves back and forth can be solved by either reducing the speeds you use (after all, it's you that is selecting the speeds) or by reducing the accellaration and jerk parameters.

Timechild2013, 2018-01-15:
Hello MB3Y, thanks for the information, it is no problem to reduce the rapid traverse and the acceleration. It only annoys me to constantly check every single layer in each part to see if Cura is having such problems. I also do not want to reduce the speed of each part in principle with the assumption that the error occurs. Some parts are 200mm high with 0.2mm layer and lasts 14 hours and it takes more than 20 hours to reduce speed. I just want to point out the bug, because it already exists from Cura 2.0.

MB3Y, 2018-01-15:
The problem in my view is not in Cura, but your printer. Each machine will have its own limits for acceleration and jerk. Just have them fix in your machine. Like that your printer HW will limit regardless what speed changes or acceleration or jerk settings will be done via Gcode.

Timechild2013, 2018-01-15:
The slicing view in Cura is not normal not because of my printer. I can reduce max acceleration to 2000 and acceleration + travel to 1000, jerk only 20 and I dont loose step, but it need much more time to print. Why should I print with Cura slowly? other slicer dont have this problem and I dont need to reduce my speed same printer.

Ghostkeeper, 2018-01-30:
So to summarise, your problem is: You're printing concentric lines at 60mm/s. To move from one circle to the next, Cura puts travel moves in between at 200mm/s. The printer loses steps at those 200mm/s.

I really suggest reducing the acceleration or jerk. Yes, your print will be slower, but it will also be more reliable. Our default travel speed is 120mm/s, not 200mm/s, but if you reduce the acceleration and jerk it will probably never reach that speed. Maybe other slicers have their default acceleration/jerk differently than Cura.

If you don't want to reduce acceleration/jerk, you need to make clear to us what Cura needs to output differently in the g-code. What's the wrong g-code and what would be the correct g-code? And why?

Timechild2013, 2018-01-30:
Hello Ghostkeeper, "To move from one circle to the next" and loose step is not correct. The problem is this micro travel step forwards and back without senseless. If you print this layer you can see what the problem is. 
This Problem is software and not hardware.
This is not correct sclicing:
https://drive.google.com/open?id=1O-0XhqIuwWh4mYnWX21u4D7iFd6ADYA7

and this is correct:
https://drive.google.com/open?id=11EhQoRP7wsrd4O6h_j5Rh0VOa8BHrosF

If you dont want to print this layer and dont see this problem you can close this issue. I really tried hard to describe the problem.

Ghostkeeper, 2018-02-21:
Ah, yes the video makes it clear.

I can't reproduce it going back and forth though. It does produce a lot of small lines with travel moves in between if the wall thickness is high enough, but it simply starts at the closest end of every line, ends at the farthest and makes a short travel move directly to the next line afterwards. No detours.

Timechild2013, 2018-02-21:
Here are 2 exampel:

https://drive.google.com/open?id=1dPdPlF-M5lbZRg79QNl9sD_etW8ABT-r

https://drive.google.com/open?id=1rwnzkGxV4GwIxRfO28gUxuYah_5LLT-D

Cura 3.2.1 profil:
https://drive.google.com/open?id=1dqCLsT4JZxUErONaIUVwy7zEPxG-VM_1

May be you can reproduce it.

and here G-Code:
https://drive.google.com/open?id=1JfqV_gAt2LzdNs4F8sJTs94JpU8SdVHB

Ghostkeeper, 2018-03-13:
I tried with the Kreis 40.1x7 model and the Prusa i3 XL using your 3.2.1 profile, but still didn't get any small movements. Perhaps I'm using a different base quality level than you?

It's usually easier to go to File -> Save Project and share that file with us. That contains a (almost) fully traceable print because it contains the model, printer set-up and any custom profiles/settings that you may have enabled. Maybe that would help us reproduce this.

Timechild2013, 2018-03-13:
Ok, than try this projekt by layer 112:
https://drive.google.com/open?id=1RXLm7dpsBShsTJMkOjSjPZYJW15XP7zl

and this by layer 510:
https://drive.google.com/open?id=1929qdQLkZfFtzAsp10lSLAszNEPAjaz4

and this smal part:
https://drive.google.com/open?id=1sry-V3lWNptrqA5yohYQqX-jQxKjY8Cq

and this smal part:
https://drive.google.com/open?id=14KEUvPoRcVbJdKQEj3MHcFSxzsVrmLbe

and if you print it, that's like driving a tank and loose step.

Ghostkeeper, 2018-04-12:
The problem here seems to be the same one as what we have been dealing with the past week with printing thin pieces. In some places it does fit one last line in between, and in some places it doesn't. A workaround is to use a slightly different line width, e.g. 0.31mm instead of 0.30mm. Because of rounding inaccuracies it doesn't fit in some places but does fit in others, even if the ring has exactly the same width everywhere.

We have a solution for that if it occurs in the printing of walls. We didn't make that solution for when printing concentric infill and I don't think that's going to be implemented soon. It increases print time a bit but doesn't influence the print quality much. And it's easy enough to avoid.

Timechild2013, 2018-04-12:
It was only some simple part of me, change to 0.31 or 0.32 or so, it doesent work same part with different diameter. If I change to 0.31 only some layer is correct but other layer with different diameter dont work. If Cura slice short lines, that is not a problem, this going back and forth though is a problem. May be Cura can fix it or not?

Timechild2013, 2018-05-04:
Ok, and how about this part with support. At layer 24-27 same issus with support.

https://drive.google.com/open?id=1r7T494PmJlje93PUphG9-ON-AhRJlApt

https://drive.google.com/open?id=1D0gLa7GIuTm4-YIV6XZWeBRDZ-9I57hF

Ghostkeeper, 2018-06-06:
We're investigating the tiny travel moves that you're seeing in the last model.

I don't think there is anything we can do about the first model. Your model is actually thicker in some parts than in others. At some point it'll start to fit another wall in there.

Ellecross, 2018-06-07:
Devs, see CURA-5455

Ghostkeeper, 2021-06-10:
This will be solved by Ultimaker/CuraEngine#1210 .

Ghostkeeper, 2021-12-14:
This is fixed now by https://github.com/Ultimaker/CuraEngine/pull/1210.

closed 2021-12-14 by Ghostkeeper