Ultimaker/Cura #3091: Cura 3.1.0 and Flashforge Creator Pro

Mike1960, 2018-01-06:
just get a Flashforge Creator Pro and liked to use Cura 3.1.0.
setting up the printer in Cura (incl. start&end-code) was done - thx to the www - and looks ok.
but printing from SD-Card is impossible. printer is heating up the bed and noozle and after reaching the set temperature nothing happend. NO print starts !
getting bloody fingers from googling about the problem.
so, shall i remove Cura from my PC and go with S3D ???

thetobby, 2018-01-06:
1. Please fill out the issue template you removed when created this issue on Github. It might seem like meaningless questions, but it is not. Any attempt goes _a long way_.
2. If you just got the printer, please confirm it works using the samples models that were included with your printer.
3. Please share the files and code and any error messages you encounter.

I fear my suggestions will not be enough to solve your problem (_=waste of time_) so I recommend that you buy a product that includes professional support like S3D.

Mike1960, 2018-01-06:
the printer works perfekt with the Flashforge Flashprint software and prints every modell correct. so , it's NOT a problem of the printer.

Ghostkeeper, 2018-01-23:
What is the start g-code you're using then? And what g-code flavour?

I've tried printing via SD card yesterday with an Ultimaker 2 and that worked perfectly. So it's either the printer hardware/firmware, the set up you're using (start g-code) or something with the g-code flavour.

Mike1960, 2018-01-23:
THX for your fast reply

problem is solved – i use another slicer now

 

by by Cura

 

Mit freundlichem Gruß

 

Dipl.-Ing.

Michael Creydt

Beratender Ingenieur

Am Steinberg 19

38315 Schladen

Tel.: 05335 905706

Fax: 05335 905703

Mobil: 015112577847

e-mail: michael@creydt-online.com

 

Web: www.creydt-online.com

E-Postbrief: michael.creydt@epost.de

Mike1960, 2018-01-23:
I am not a debugger, I need a working and running 3D- printer 😉

 

Mit freundlichem Gruß

 

Dipl.-Ing.

Michael Creydt

Beratender Ingenieur

Am Steinberg 19

38315 Schladen

Tel.: 05335 905706

Fax: 05335 905703

Mobil: 015112577847

e-mail: michael@creydt-online.com

 

Web: www.creydt-online.com

E-Postbrief: michael.creydt@epost.de

 

That doesn't sound like the bug is solved 😛

ChrisTerBeke, 2018-01-23:
The general rule is that both Cura and the printer support the Marlin/RepRap standard. If there's an issue with compatibility it's either a Cura bug, a wrong start/end G-code or a bug in the printer firmware. Given that many people use Cura successfully with other 3rd party printers, I assume it's the configuration or a firmware bug. Given that another slice (which one?) works fine with your printer, we could rule out a firmware bug. So that leaves start/end gcode configuration. Can you give a project file or more details about your printer configuration in Cura so we can have a look?

TheOneEddy, 2018-01-30:
I have somewhat successfully got Cura to work on my flashforge creator pro
This is what i did to get it to work:
 In Cura goto Settings->Printer->Manage Printers...
 Press Add->Custom->Custom FDM printer,
Type "Flashforge Creator Pro" under "Printer Name", press OK

Set:
    X = 225
    Y = 145
    Z = 150

Check "Origin at center"
    Check "Heated bed"
   Gcode flavor "Marin"

Number of Extruders "2"
    Material diameter "1.75"
    Nozzle size "0.4" (under  "Extruder 1 and 2")
  
  Paste the following into "Start Gcode":

        M136 ; start build
        M73 P1
        G90 ; absolute coordinates
        M140 S{print_bed_temperature}
        M104 S{print_temperature} T0
        M104 S{print_temperature} T1
        G130 X118 Y118 A118 B118 ; set stepper motor Vref to defaults
        T0 ; home on the right nozzle
        G28 X Y Z ; home all axes at homing speed
        G92 X0 Y0 Z0 A0 B0 ; set all coords to 0 for now
        G1 Z5 F500 ; move Z 5mm away so we can carefully hit the limit switch
        G161 Z F100 ; home Z slowly
        M132 X Y Z ; recall stored home offsets for XYZ axes
        G1 X110 Y-72 Z30 F3300 ; move to waiting position
        M116 ; wait for temps
        G92 E0 ; set current extruder position as 0 so that E15 below makes sense
        G1 X110 Y-70 Z0.2 F2400.0 ; move to just on the bed
        G1 X110 Y70 E15 F1200.000 ; extrude a line of filament along the right edge of the bed
        G92 E0 ; set E to 0 again because the slicer's next extrusion is relative to this
        T1;
        G92 E0 ; set current extruder position as 0 so that E15 below makes sense
        G1 X105 Y-70 Z0.2 F2400.0 ; move to just on the bed
        G1 X105 Y70 E15 F1200.000 ; extrude a line of filament along the right edge of the bed
        G92 E0 ; set E to 0 again because the slicer's next extrusion is relative to this
        T0;

   Paste the following into "End Gcode":

        M73 P100 ; end build progress
        M140 S0 ; set bed temperature to 0
        M104 S0 T0 ; set extruder temperature to 0
        M104 S0 T1 ; set other extruder temperature to 0
        G1 Z150 F1100 ; send the build plate to the bottom
        T0 ; Next job assumes T0 is the current tool
        G28 X Y F2500 ; home X and Y axes
        M18 ; disable all stepper motors
        M70 P5 (We <3 Making Things!)
        M72 P1 ; Play Ta-Da song

Press "Close"

Flashforge does not support gcode and therefore the export files must be converted to x3g

Click "Plugins" -> "Browse Plugins..."
Find"X3G Writer" and press "Download"
Restart Cura

I got some medium good print with this setup, but I have issues with under extruding in the beginning of each layer. Does anyone have a solution for this?

Ghostkeeper, 2018-02-21:
Maybe you have "Retract at Layer Change" enabled? Does it improve if you change that?

Otherwise you should take a look at your line widths. Some printers are very sensitive to flow changes, so if you use different line widths for your walls or something you could get over- or underextrusion due to the change in material flow speed.

steverc1572, 2018-03-31:
I believe his problem is Flashforge Creator Pro uses Sailfish or least some of the early models did. I used to have one too and never could get it to work to my satisfaction. So is there a way for Cura to work with Sailfish?

steverc1572, 2018-03-31:
Since Ghostkeeper mentioned it, How do you turn a good profile into a full fledge printer profile ?

thawkins, 2018-03-31:
I think you need to set the gcode flavor to "makerbot" and not leave it on Marlin. The FFCP uses sailfish which is a clone of the system used in the original Makerbot 1 Dual machines, not it is NOT a clone of the makerbot 2 dual which most people seem to chose, so make sure your x3g plugin is set for the m1d

thawkins, 2018-03-31:
This is my start script from my s3d system you may need to change it to plug in the values in the [....] brackets. 

; **** Replicator 1 dual start.gcode ****
M73 P0 ; enable build progress
G162 X Y F3000 ; home XY maximum
G161 Z F1200 ; home Z minimum
G92 Z-5 ; set Z to -5
G1 Z0 ; move Z to 0
G161 Z F100 ; home Z slowly
M132 X Y Z A B ; recall home offsets
M135 T0 ; load right extruder offsets
G1 X150 Y-70 Z30 F9000 ; move to wait position right hand side of the table
G130 X20 Y20 Z20 A20 B20 ; lower stepper Vrefs while heating
M126 S[fan_speed_pwm]
M140 S[bed0_temperature] T0
M134 T0 ; stabilize bed temperature
M104 S[extruder0_temperature] T0
M133 T0 ; stabilize right extruder temperature
G130 X127 Y127 Z40 A127 B127 ; default stepper Vrefs
G92 A0 B0 ; zero extruders
G1 Z0.4 ; position nozzle
G1 X-115 Y-80 Z0.15 F1200 ; slow wipe
G1 X-110 Y-80 Z0.5 F1200 ; lift
G1 X100 Y-83 F14000.000; move to front right corner of bed
G1 X-90 Y-83 E24 F2000.000; extrude a line of filament across the front edge of the bed
G92 A0 B0 ; zero extruders
M73 P1 ;@body (notify GPX body has started)
; **** end of start.gcode ****

pcollingwood, 2018-03-31:
Similar issue, but no file is being created. My setup:
Win 10
Cura 3.2.1
FlashForge Creator Pro (Info from screen)
-Sailfish Creator Pro
-Free SRAM  830
-ATMega 2560 16/05/23
-Sailfish v7.8 r0e8a9
(Not that the machine info matters, it's not getting to there...!)
When slicing and then trying to save I get the pop-up attached.
[image]

Any ideas what I need to do to rectify this?
It did work with 3.2.0 to a degree, but always put the model off the print bed which then needed to be messed about with to place it correctly. It was as if the origin had been set to centre instead of what I actually had in the software.

thawkins, 2018-04-01:
On a makerbot/flashforge sailfish based printer the origins IS in the
center of the bed.

pcollingwood, 2018-04-01:
Oh! For some reason I thought it was front left. I've changed that now. Unfortunately it has not helped with the error/warning message. I still get that and a 0 byte file.
Any ideas?

pcollingwood, 2018-04-01:
I also only get one option for Makerbot in the drop down menu. Should I be looking somewhere else?
[image]

pcollingwood, 2018-04-01:
Back to the error message - It has been resolved, but the method I used does not make a whole heap of sense.
I renamed the "X3GWriter-master" to "X3GWriter-master Old". Then copied the "X3GWriter-master" folder from the remnants of the Cura 3.1.0 folder into the Cura 3.2.1\plugins folder and it still didn't work.
I then deleted the copy of "X3GWriter-master" that I had just pasted into the plugins folder and renamed "X3GWriter-master old" back to "X3GWriter-master".
I loaded a file (the same one I've been using all along) and followed the same process, but this time I got the DOS text screen and it created the file. I loaded it into the FlashForge program to look at it and it's where it should be, how it should be.
Confusing!
By the way, where is the output (that appears in the DOS window) stored? There were lines that had the word "error" in them, but it's gone too quick to read.

closed 2018-04-01 by Mike1960

Ghostkeeper, 2018-04-26:
The X3GWriter plug-in should put the stderr output of the GPX application into Cura's log. At least, the latest version does. You should be able to find it by going in Cura into Help -> Show Configuration Folder. There should be a cura.log file somewhere there.

Ghostkeeper, 2018-04-26:
> Since Ghostkeeper mentioned it, How do you turn a good profile into a full-fledged printer definition file?

You need to get a text editor and open up your profile file in Cura's configuration folder. Then also find one of the existing printer definitions (.def.json files in Cura's installation folder, like in Program Files). I suggest you copy one of those printer definitions and fill in the values for your own printer. That's how I always do it.

steverc1572, 2018-04-27:
Thanks, Ghostkeeper that's exactly what i did to make one for Formbot raptor.

Ghostkeeper, 2018-05-25:
> Thanks, Ghostkeeper that's exactly what i did to make one for Formbot raptor.

If you have a nice file, consider submitting it to Cura via a pull request :wink:

steverc1572, 2018-07-04:
The problem with the FlashForge Creator Pro and KNP3D is the M109 command, it tells the bed to set build platform temperature in degrees Celsuis and the M190 command. I can do post processing and tell it to replace the M190 with M109 but I have a extra M109 already in the gcode telling it to wait for the Nozzle to reach temp. I just manually go int and remove the M109 command for the nozzle.

Here is a example of the untouched gcode generate by Cura for Marlin ( Makerbot does the same )
[image] 

 
 Now here is it processed with

[image]

[image]

 so I have to manually remove this or my bed will go to the set number

[image]

Ghostkeeper, 2018-08-15:
I suggest trying the Makerbot flavour.

steverc1572, 2018-08-16:
Makerbot flavor does the exact same thing. There needs to be a sailfish flavor. 

;FLAVOR:Makerbot
;TIME:25982
;Generated with Cura_SteamEngine 3.2.1
M190 S60
M104 S210
M109 S210
M82 ;absolute extrusion mode
(**** Start.gcode ****)
M103 (RPM off)
M73 P0 (enable build progress)
G21 (set units to mm)
G90 (set positioning to absolute)
(**** begin homing ****)
G162 X Y F2500 (home XY axes maximum
G161 Z F1100 (home Z axis minimum)
G92 Z-5 (set Z to -5)
G1 Z0.0 (move Z to "0")
G161 Z F100 (home Z axis minimum)
M132 X Y Z A B (Recall stored home offsets for XYZAB axis)
(**** end homing ****)
G1 X-110.5 Y-74 Z150 F3300.0 (move to waiting position)
G130 X0 Y0 A0 B0 (Lower stepper Vrefs while heating)
M6 T0 (wait for toolhead, and HBP to reach temperature)
G130 X127 Y127 A127 B127 (Set Stepper motor Vref to defaults)
M108 R3.0 T0
G0 X-110.5 Y-74 (Position Nozzle)
G0 Z0.6      (Position Height)
M108 R4.0    (Set Extruder Speed)
M101         (Start Extruder)
G4 P1500     (Create Anchor)
(**** end of Start.gcode ****)

steverc1572, 2018-09-16:
The problem was with the KNP3D converter, when I used the X3G converter plugin from Cura it worked like a champ. So if anyone is trying to use a KNP3D printer with Cura just install the plugin form the menu and ditch the one with the KNP3D.

SkydiveBadger, 2020-09-25:
I'm also trying the FlashForge Creator PRO with Cura... But I'm ready to pull my hair out with the homebrew workaround/fix.

I'm still generating totally empty (zero bytes) files....  @Ghostkeeper have you got any new advice or options to attempt?

Ghostkeeper, 2020-09-28:
Please keep these discussions to the X3GWriter repository. There are existing issues there.

steveb, 2020-09-28:
I use https://cloud.astroprint.com/ to generate FlashForge Creator PRO files now, which I think uses Cura under the hood.

Ghostkeeper, 2020-10-02:
No, X3GWriter uses [GPX](https://github.com/markwal/GPX) under the hood.