Ultimaker/Cura #3092: Combing doesn't works correctly with walls>1

MB3Y, 2018-01-06:
**Problem description**
Slicing any regular standing up litho (slightly curved as well as cylindrical models) with Combing=all gives me the expected behavior only when selecting Wall Line Count = 1. I.e: all travel moves stay strictly inside the model.
When I select Wall Line Count 2 or more, many travel moves are made outside the model.
Minimum model thickness = 0.4, my nozzle extrude size.
I tried to change many parameters to get Combing behave correctly with multiple walls, but all in vain.

**Application Version**
Cura 3.1, but also noticed on previous versions

**Platform**
Windows 10

smartavionics, 2018-01-06:
Perhaps it's the same as this https://github.com/Ultimaker/CuraEngine/issues/625 ?

MB3Y, 2018-01-06:
In lithos you certainly have many/much more thin walls compared to other prints. So if the bug is related to hitting one of the combing boundaries, then you have found a simple reproduction scenario.
I just tested and see that if I do increase the minimal thickness of the litho, then this combing error improves and eventually disappears - but for a litho that is not really an option...

Ghostkeeper, 2018-01-23:
I'm trying this with a hollow cylinder (thickness 0.8mm nor thickness 1.6mm), but didn't get any moves outside. But the bug in Ultimaker/CuraEngine#625 occurs only in edge cases, so it could very well be that it would occur if the lines happen to fall in the wrong place. If that's the case, I'd like to close this as duplicate of Ultimaker/CuraEngine#625.

Is it possible to share one of the models you tried?

MB3Y, 2018-01-23:
Glad to share.
ZIP contains

1. the stl
2. my CuraProfile
3. picture of result slicing with WallCount=1
4. picture of result slicing with WallCount=2

[CombingNOKWithWallCountAbove1-LithoExample.zip](https://github.com/Ultimaker/Cura/files/1657456/CombingNOKWithWallCountAbove1-LithoExample.zip)

jackha, 2018-01-24:
I can confirm that this model has the combing problem, even with the default settings, while a "generic" hollow cylinder does not.
 
[image]

Ghostkeeper, 2018-02-12:
It's not an edge case. I've found that there are 2 reasons why it's doing this:

* In Cura, the walls are normally drawn from inside to outside (or from outside to inside if "Outer Before Inner Walls" is enabled). So it draws all inner walls first, then the outer wall. Because the lithography has a lot of dents, the inner walls are discontinuous. So it needs to move to another part to draw the inner wall there.
* Combing has a distance to keep away from the outer walls when moving around or within them: Travel Avoid Distance. It is unable to maintain that distance in such a thin solid. But even when you set that distance to 0, the walls of opposite sides are still touching so it can't squeeze between them.

If there is only the outer wall, the walls are no longer touching so any of the skin lines in the irregular layers are able to squeeze between the walls and it combs correctly. What's more, the parts are printed more localised because the skin and fill-small-gaps lines are quite local themselves, so the nozzle doesn't need to move around as much.

kkava, 2018-02-12:
This shouldn't be closed. None of the three combing settings lead to the head staying inside of the part while printing in C3.2.1. This is leading to terrible surface quality on my prints.
[GI3X_hook_part0-export.zip](https://github.com/Ultimaker/Cura/files/1716326/GI3X_hook_part0-export.zip)

MB3Y, 2018-02-12:
@Ghostkeeper : I can understand the reasoning based on the current implementation of Combing, but I do agree with @kkava that this item should not be closed.
First of all we should agree on the advantages of combing.

- Is it only speed (gaining time with less retracts - but loosing in longer travels)
- Or is it also less stringing and oozing?!

- If it is solely the first, then one can still argue that shorter retracts are better - and one can still improve that (e.g. combing now can jump from one end of a letter C to the other end of the letter C over the shortest void between them - while it is more interesting to travel over the lines of that letter C and take the shortest bridge between the different combing areas, and cross there.
- If it is also the latter then definitely some effort should  be done 

I'd like to use multiple walls, while keeping the combing advantage as if I did set one wall.

CoryJRC, 2018-03-04:
+1 for reopening this request.  Combing is such a useful feature for keeping prints clean, but the fact that it does not have this effect when walls>1 renders it useless for many prints.

Attached classic octopus print that.

Please consider re-looking at this.

Thank you.
[image]

kkava, 2018-03-08:
It is 100% stringing and oozing-related for me.

MB3Y, 2018-04-03:
Hi @Ghostkeeper,
That's were I suggested #3458. When there is no combing path **not** hitting a wall, then you have no other option then to hit a wall. Just be picky in such case rather than just taking the shortest route to the next point.
Without me rereading the #3458 ticket again:
Rule1: Try to find a travel path that is not **crossing** walls, i.e. allow moving over existing walls (or places where lateron a wall will be printed) or move over the interior side of touching walls.
Rule2: Cross from one valid combing area to the next on the point with shortest distance between both. 

For both rules use standard retraction over the  whole length of the travel (unless disabled of course). So do not retract only when crossing the void between the valid combing areas... Like this oozing and stringing will be less or done by the time you cross and possibly also the negatif effect of the oozing when crossing the wall.

I think it's worth a try?

MB3Y, 2018-04-26:
@Ghostkeeper and colleagues: It's a joy to see that you Ultimaker/Cura guys are taking the community input serious and providing understandable and respectful feedback.
It makes me smile (and silently a bit proud too) when I read the release notes and find bugs solved where also my/the community's input and reasoning was clearly heard and taken into account.
Way to go! Keep up the good work!

jackha, 2018-05-02:
I've been working on this issue for a while now and I'm glad to say that it got better ^^. https://github.com/Ultimaker/CuraEngine/pull/754

@MB3Y The model you provided is a very good test object because the wall can be as thin as a single line and it can be quite thick as well. 

This is the result I got (no travels on the inside of the cylinder):
[image]

Ghostkeeper, 2018-05-28:
I think we can consider this fixed (in version 3.4) then, with jackha's implementation of the 3 rules I put up above.

closed 2018-05-28 by Ghostkeeper