Ultimaker/Cura #3093: Cura 3 printing part before ooze shield

xile6, 2018-01-06:
I can not get cura to print the ooze shield before printing the item.
On a tool swith the nozzle will not prime because cura will print the part then the ooze shield.
I.e.

Nozzle 1 prints ooze shield, then part.
Nozzle 2 will print part then ooze shield.
Noozle 1 will print part then ooze shield.
This leaves the noozle not prime and holes in the parts.

I tried to use the build in prime tower but the order is the same. 
Part, prime tower, ooze shield.
Leaving holes in the part or fail prints

nallath, 2018-01-10:
This is intentional. The ooze shield is not meant to prime. The prime tower is. I just tried a print, but with me i do get the prime tower being printed before the part.

xile6, 2018-01-10:
Yes i see that it will do the prime tower before part.
Is the any way to get it to do the ooze shield first?
I use S3D for my dual prints and it does the ooze shield first and i dont need to use a prime tower.
But cura has better support settings so wants to give it a try.

Ghostkeeper, 2018-01-25:
Indeed if you use the ooze shield before the print then you may not need a prime tower (depending on your material).

I fear a bit for when there is a lot of oozing and the nozzle starts with the ooze shield at the same spot on every layer. In the prime tower we added a clever golden-ratio spreading function so that it would not wipe off the ooze at the same spot every time, because that bunches up and causes the nozzle to push against the tower and break it.

Ghostkeeper, 2018-01-25:
I can't reproduce this. The ooze shield seems to be printed first every time.

Here's the project file I'm testing with (rename to UM3_ooze_shield_test.curaproject.3mf):
[UM3_ooze_shield_test.curaproject.zip](https://github.com/Ultimaker/Cura/files/1664066/UM3_ooze_shield_test.curaproject.zip)

xile6, 2018-01-25:
I have a video that i pisted to the cura fourm

https://youtu.be/XixAlVk9Rjc

I will check out youd file.
But i tired everything and the ooze shield still prints lasts

Ghostkeeper, 2018-02-13:
Also in your video it prints the ooze shield first on the new layer. It appears like it prints the model first but in fact that is on the lower layer. When it switches to the next layer the ooze shield is printed first.

There is no way to print the ooze shield of the next layer before printing the part of the previous layer, because then the nozzle would have to move back down and we can't guarantee that your nozzle won't hit the ooze shield when it needs to move to other parts.

xile6, 2018-02-13:
No it is doing the part then ooze shield as in color.

You might see the the video doing the shield first, but thats kn the last color it used. Which means

Color 1 ooze shield
Then
Color 2 part
Color 2 ooze
Color 1 part.

It should be color 2 ooze
Color 2 part
Color 1 ooze
Color 1 part

Ooze of that color before part of that color.

xile6, 2018-02-13:
I will get a video of S3D which does the shield before part and compare.

Ghostkeeper, 2018-03-07:
I think I understand what you mean. Currently the ooze shield is always printed first on a layer. But you want it to be printed first after an extruder switch.

How would this behave when you have 3 colours? Can you think of an order in which to print this then (and include the layer switches maybe)? I don't see how this could work then.

ianpaschal, 2018-03-26:
It's working as intended and OP didn't respond for more than a month. Closing.

closed 2018-03-26 by ianpaschal