Ultimaker/Cura #3095: Insufficient Nozzle Size Choice

Gavski, 2018-01-07:
Trying Cura 3.1 - want to move from 15.04.6.
I have UM2 selected with Olsen Block Mod.
Mac OSX platform

I can only select 1 of 4 predefined nozzle sizes on GUI or Menu Settings/Nozzle.

With Olsen block one can install any size nozzle - for example I have 0.3mm in there now.  This is not on the pre set list.

Old version had arbitrary choice.

Please reinstate this feature or add more options matching what the market supplies.

Cura 3.1 is therefore unusable with my configuration.

Thx
J

filipgoc, 2018-01-07:
@Gavski This is not a fix, but a quick workaround: you can simply use the 'line width' (extrusion width) property to slice for whatever nozzle you desire. For 0.3 I'd go for 0.33 maybe for starters.

Renha, 2018-01-08:
same here - built new machine, was trying to figure out bad calibration cubes, it was Cura trying to print 0.4 mm lines with 1.2 mm nozzle, claiming that this value was carefully calculated from nozzle diameter.

nallath, 2018-01-10:
The things that you can select are actually those that we have pre-defined profiles for. So selecting the dropdown does way more than what old cura would do. 
So if you want to change it; you can use line width.

closed 2018-03-26 by ianpaschal