Ultimaker/Cura #3096: master branch - model position coords now relative (bad news)

smartavionics, 2018-01-07:
**Application Version**

master branch

**Platform**

Linux

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**

Add model to build plate, select it and move tool, enter (x/y/z) values.

**Actual Results**

Entered values are interpreted as deltas rather than as absolute (as it used to be). Also when the field loses focus, the model shifts again as the delta is applied again.

**Expected results**

Values in the move tool field should be absolute.

**Additional Information**

I can see that being able to apply relative movements is good but this change is a nightmare to use right now. For example, if when the model is imported/scaled/whatever it doesn't sit on the build plate because z was non-zero, you could set z to zero and the model would sit on the plate. Now, if the model is not sitting on the plate one can only guess as to what the z offset to get it to the plate will be.

fieldOfView, 2018-01-07:
This must be an unintended change, possibly caused by the multi build plate work?

smartavionics, 2018-01-07:
Maybe. I have to say that it pisses me off how often the devs commit stuff to the master branch that obviously hasn't been well tested. If it's WIP, why can't they merge it when it's all working sufficiently well that it doesn't screw other people.

fieldOfView, 2018-01-07:
I don't disagree.

One of the problems is that the testing team can't test stuff unless it is committed in one of the main branches. They test builds, and only the main branches are built.

smartavionics, 2018-01-07:
> One of the problems is that the testing team can't test stuff unless it is committed in one of the main branches. They test builds, and only the main branches are built.

Hmm, they really need to have a dev branch built and merge into that before wrecking the master branch. It looks to me that the 3rd party code that is contributed by people like myself gets tested a lot better before it gets into the master branch than the code that comes from the cura devs.

I am realistic and do realise that the odd mistake can slip through, we're all human (aren't we?)

fieldOfView, 2018-01-07:
The idea is that master is that dev branch.

(I am not too sure @ghostkeeper is actually human; humans need sleep)

ChrisTerBeke, 2018-01-07:
Master is absolutely not mean as stable. Everything that's not an official release is used for development. Any stable release is tagged, so if you want to use Cura for actual usage, just use the latest stable (or try to live with the occasional bug that appears in master).

smartavionics, 2018-01-07:
I understand your point about using a tagged release. But as a coder, I don't want to use a tagged release, I want to use the latest code because that's what I'm coding against. If master isn't meant as stable why do contributors have to wait months for their PRs to be merged? It's either reasonably stable and code is vetted/tested before being merged or it's not stable and code is merged sooner and everyone lives with the odd glitch. You can't have it both ways.

ChrisTerBeke, 2018-01-07:
We try to test it as much as possible before merging, but our internal QA uses master branch for final testing because that's what the nightly builds are created from. This means that sometimes a bug slips into master, but not that it's unusable to code against imo.

smartavionics, 2018-01-11:
This appears to have been fixed now, thanks. Closing...

closed 2018-01-11 by smartavionics