Ultimaker/Cura #3098: Cura forces compatibility mode even though disabled in settings

mitgobla, 2018-01-07:
**Application Version**
3.1.0

**Platform**
Windows
Using version from Ultimaker's website

**Display Driver**
Intel HD Graphics 3000

**Steps to Reproduce**
Load in model, go to layer view. Using custom printer settings (Wanhao I3 V2) that uses Marlin.
Ran twice with and without administrator.

**Actual Results**
It says it's in compatibility mode and there's no option to simulate the print head or preview feedrate visualization. In the settings the option 'Force layer view compatibility mode' is disabled.

**Expected results**
Layer view should show options for feedrate visualization and print head simulation

**Additional Information**
Screenshot of the layer view
[image]

ChrisTerBeke, 2018-01-08:
Can anyone reproduce this? I can't...

awhiemstra, 2018-01-09:
The relevant information is in:

> Display Driver
> Intel HD Graphics 3000

According to https://en.wikipedia.org/wiki/List_of_Intel_graphics_processing_units#Sixth_generation , the maximum supported OpenGL version on Windows is 3.0. We require OpenGL 4.1 for the non-compatibility mode.

ChrisTerBeke, 2018-01-09:
In that case it's not a bug, closing issue.

closed 2018-01-09 by ChrisTerBeke

tlhintoq, 2018-07-15:
@awhiemstra @ChrisTerBeke 

Am I interpreting this right then?  The Windows version of Cura will therefore never be on par with say the Mac version _because of_ OpenGL?

All the other apps in the world do amazing graphics... The 3d design software like Autocad and Solidworks, not to mention all the games out there... Can all do awesome graphics in real time... But for rendering the slicing there's nothing y'all can do to use another rendering technology on Windows?  Thus Cura for Windows will just keep dropping further and further behind?

Is that the take away from this?

ChrisTerBeke, 2018-07-15:
Eeehm no? We just require OpenGL 4.1 or higher, regardless of OS. I don't see an issue with this as most laptops/desktops of 5 years and older have cards that support this. Intel HD 4000 and up seems to support this. We can see in our anonymous analytics data that hardly any users are on platforms that don't support 4.1, so that's why it's not a priority for us. If anyone in the community wants to put in effort, that's always welcome for us.

ChrisTerBeke, 2018-07-18:
I think he meant 3.0 regarding the Intel HD Graphics 3000 (at least, my comment was based on that, not on Windows as a whole).

Tomaskom-cz, 2018-10-15:
I think the real issue for most people is not the compatibility mode with simplified graphics itself, but that compatibility mode does not have the print head movement simulation slider, forcing users to export the gcode and use external tools (like http://gcode.ws) for very crucial checks, slowing down workflow and killing comfort. 

I can't see a reason (other than low/zero priority for allocating devs time on the feature) why this couldn't be implemented quite easily. 
When it comes to hardware requirements, rendering less lines with some simple marker for the print head would be similar or less demanding compared to rendering the full layer, which is already being done anyway. So not being on OpenGL 4.1 is not an issue. 

Should I start this as a feature request?

fieldOfView, 2018-10-15:
> Should I start this as a feature request?

If it is quite easy, you should start implementing it, and make a pull request once you're done.

samtaylor3d, 2019-03-16:
im having the same problems and im not understanding if theres something i can install on my i5 laptop to get regular layer view working. i looked into open gl but it seems like a programming suite not a driver. what am i missing and should i just install open gl4.1 and hope for the best? or is it not that simple?

fieldOfView, 2019-03-17:
Your _GPU_ has to support openGL 4.1. If it doesn't, there is nothing you can install that will let you use the "full" layer view (expect a better GPU, but that is infeasible with a laptop).

What i5 do you have? Intel has been using that name for a long time now. Old i5 processors have integrated graphics that don't support openGL 4.1. Newer i5 processors have integrated graphics that do. Note that openGL 4.1 is a ~9 year old standard.

tlhintoq, 2019-03-18:
If your PC has another display adapter as most do, such as NVIdia - double check the control panel for that adapter.  You can specify that the other card should be the default use for Cura.

That was the case for one of my laptops: I had to specify using the NVidia and then make sure NVidea was set to use full 3D processing.

erkserkserks, 2021-12-20:
For people coming here via a search engine:

It's possible to use a software OpenGL renderer as a workaround. On Windows, you can use llvmpipe from here: https://github.com/pal1000/mesa-dist-win

Unzip a release, e.g. mesa3d-21.3.2-release-mingw.7z, and run `perappdeploy.cmd`. It will ask where to install the software renderer to, point it to the Cura installation directory, e.g. `C:\Program Files\Ultimaker Cura 4.12.1`

My laptop's GPU (Intel HD Graphics 3000) only supports OpenGL 3.1, but I was able to use the software renderer to unlock all of Cura's rendering modes (layer thickness, speed, flow). Performance is pretty bad (~1 FPS), but it is at least possible to use these features.

rociomarielgnas, 2023-05-26:
> For people coming here via a search engine:
>
> It's possible to use a software OpenGL renderer as a workaround. On Windows, you can use llvmpipe from here: https://github.com/pal1000/mesa-dist-win [...]

Coming here via search engine indeed. Thank you for an actual useful answer instead of snappy remarks like the devs gave. You're the mvp.

TheKerevas, 2024-09-16:
> For people coming here via a search engine:
>
> It's possible to use a software OpenGL renderer as a workaround. On Windows, you can use llvmpipe from here: https://github.com/pal1000/mesa-dist-win [...]

I have a Dell laptop with an Intel HD 4000, I want to try this method but I can't, what exactly should I do?