Ultimaker/Cura #3099: Cura 3.1.0 - Minor issues when duplicating a material

combs, 2018-01-07:
**Application Version**

3.1.0

**Platform**

OS X 10.11.4

**Qt**

**PyQt**

**Display Driver**

Seeing this on several machines
**Steps to Reproduce**

OK! This is not big stuff. Just minor UI.

_Wrong material selected after a duplication_
- Choose Settings->Material->Manage Materials...
- Click a material on the left
- Press the Duplicate button
- **Actual results:** The last material in the list is selected (e.g. Verbatim/BVOH on my machine)
- **Expected results:** The newly duplicated material is selected

_Filament color changes do not appear for newly duplicated filament_
- Choose Settings->Material->Manage Materials...
- Click a material on the left
- Press the Duplicate button
- Make changes to the filament color. click Close
- **Actual results:** The new material appears in the main Cura window's Material dropdown, but filament color change is not reflected. 
- **Expected results:** The new material is selected and all properties are reflected. 

I have not exhaustively tested all properties but diameter changes do appear correctly right away. 

**Additional Information**

ChrisTerBeke, 2018-01-08:
The first issue I can't reproduce, for me it selects the new material after duplicating (which is editable, the original one is not).

For the 2nd issue I made a new ticket to get it fixed. Devs: CURA-4774.

ChrisTerBeke, 2018-01-09:
Nr 2 is fixed in https://github.com/Ultimaker/Cura/commit/f9286d4bd59a29ec6c33d3345d5c7c18db53027d.

ChrisTerBeke, 2018-01-09:
I can now reproduce nr 1 as well and making a fix for it.

LipuFei, 2018-01-09:
Nr 1 is fixed in 93599de64239644ea8a86434d5f2bbafa2b47873

combs, 2018-01-09:
y'all rule, thanks very much 🎉

closed 2018-01-09 by combs