Ultimaker/Cura #3100: Cura 3.1.0 - Color picker can appear under Materials window on OS X

combs, 2018-01-07:
**Application Version**

3.1.0

**Platform**

10.11.4 and 10.12.6

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**

- Choose Settings->Material->Manage Materials...
- Click a material on the left
- Attempt to alter its color

**Actual Results**

- The color picker appears but is not the foremost window. If its default placement overlaps the Materials dialog, user may be mildly confused. Moving the Materials dialog reveals the missing color picker

**Expected results**

- The color picker appears and is the foremost window. 

**Additional Information**

ChrisTerBeke, 2018-01-08:
For me it's always the top window (on Windows and MacOS). Can you send a screenshot of what's happening maybe?

combs, 2018-01-09:
I will grab a screenshot when I can (sorry, on wrong machine right now). But it won't be a surprising screenshot :) Just shows the color picker as a background window.
I'll play with it some more and see if a logout/restart changes anything.

combs, 2018-01-11:
[image]

Restarting didn't change. Seeing it on two machines with different OS versions (not a cloned workstation setup)

ChrisTerBeke, 2018-01-11:
Mm interesting, can't reproduce on master branch currently. Do you mind testing it in the 3.2 beta that comes out in ~1.5 weeks?

Ghostkeeper, 2018-01-28:
It still seems to work all right on 3.2 on Windows, but I can't test with MacOS.

ianpaschal, 2018-03-27:
This still reproduces on macOS in master.

closed 2023-06-24 by github-actions[bot]