Ultimaker/Cura #3101: Cura 3.1.0 - Invisible unused settings errors I shouldn't care about

Renha, 2018-01-08:
**Application Version**
3.1.0

**Platform**
Windows 10

**Steps to Reproduce**
Add Minimum layer time to visible settings, while not adding Prime tower and/or anything about fans (imagine you doesn't have any, like I am).
Set Minimum layer time to 20 seconds

**Actual Results**
Slicing fails with errors in settings Prime Tower X Position, Prime Tower Y Position, and Regular/Maximum Fan Speed Threshold O_o

**Expected results**
Errors about things I don't care about and don't even have (!) doesn't annoy me and doesn't affect slicing.

**Additional Information**
[image]

ChrisTerBeke, 2018-01-08:
I can't reproduce this using the default settings (+ your changes) on an Ultimaker 3 or Ultimaker 2+. Can you tell a bit more about the printer you have configured? Maybe share a project file that contains these settings?

Renha, 2018-01-08:
no problem, there are machine settings:
[image]
this is the project with error (I've added  Regular/Maximum Fan Speed Threshold to visible, setting it to 100 'fixes' problem for that exact case).
[AMKI_xyzCalibration_cube.curaproject.3mf.zip](https://github.com/Ultimaker/Cura/files/1613322/AMKI_xyzCalibration_cube.curaproject.3mf.zip)

Ghostkeeper, 2018-01-23:
Hmm, that project gives an error when loading it in version 3.2:

    2018-01-23 14:26:07,353 - WARNING - UM.Logger.logException [76]: Exception: We failed to serialize the stack. Trying to clean up.
    2018-01-23 14:26:07,355 - WARNING - UM.Logger.logException [80]: Traceback (most recent call last):
    2018-01-23 14:26:07,355 - WARNING - UM.Logger.logException [80]:   File "/home/ruben/Projects/Cura/cura/../plugins/3MFReader/ThreeMFWorkspaceReader.py", line 772, in read
    2018-01-23 14:26:07,356 - WARNING - UM.Logger.logException [80]:     stack.deserialize(extruder_file_content, file_name = extruder_stack_file)
    2018-01-23 14:26:07,356 - WARNING - UM.Logger.logException [80]:   File "/home/ruben/Projects/Cura/cura/Settings/ExtruderStack.py", line 153, in deserialize
    2018-01-23 14:26:07,356 - WARNING - UM.Logger.logException [80]:     self.setNextStack(stacks[0])
    2018-01-23 14:26:07,356 - WARNING - UM.Logger.logException [80]:   File "/home/ruben/Projects/Cura/cura/Settings/ExtruderStack.py", line 87, in setNextStack
    2018-01-23 14:26:07,356 - WARNING - UM.Logger.logException [80]:     Application.getInstance().getExtruderManager().updateMaterialForDiameter(position)
    2018-01-23 14:26:07,357 - WARNING - UM.Logger.logException [80]:   File "/home/ruben/Projects/Cura/cura/Settings/ExtruderManager.py", line 514, in updateMaterialForDiameter
    2018-01-23 14:26:07,357 - WARNING - UM.Logger.logException [80]:     extruder_stack = global_stack.extruders[str(extruder_position)]
    2018-01-23 14:26:07,357 - WARNING - UM.Logger.logException [80]: KeyError: '3'

ChrisTerBeke, 2018-05-23:
No new info, cannot reproduce. Closing.

closed 2018-05-23 by ChrisTerBeke