Ultimaker/Cura #3102: Using GPX to prep GCode files gives error

Spinnergee, 2018-01-08:
I've been using GPX to recode the Cura GCode files for use with a Replicator - Until about a week ago it has been going swimmingly, but now every time I try to create an X3G file using the same process, it gives this error:

(line 14) Semantic warning: Z axis homing to maximum endstop
(line 21) Syntax warning: unrecognised command word 'i'
(line 21) Syntax warning: unrecognised command word 'n'
(line 21) Syntax warning: unrecognised command word 'i'
(line 21) Syntax warning: unrecognised command word 'n'
(line 21) Syntax error: unrecognised gcode '..
'
(line 57509) Buffer overflow: input exceeds 1024 character limit, remaining characters in line will be ignored

**Application Version**
15.04.4

**Platform**
Windows 10 Home (17)

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**
1) Create a GCode file in Cura 
2) Run GPX  (GPX -m r1 <insert Filename here>

**Actual Results**
(line 14) Semantic warning: Z axis homing to maximum endstop
(line 21) Syntax warning: unrecognised command word 'i'
(line 21) Syntax warning: unrecognised command word 'n'
(line 21) Syntax warning: unrecognised command word 'i'
(line 21) Syntax warning: unrecognised command word 'n'
(line 21) Syntax error: unrecognised gcode '..
'
(line 57509) Buffer overflow: input exceeds 1024 character limit, remaining characters in line will be ignored

**Expected results**
X3G file to be created, as it has for years.

**Additional Information**

ChrisTerBeke, 2018-01-08:
The error doesn't appear to be in Cura, but in your GPX post processor? Did you update Cura or GPX? The Cura output hasn't changed in a long time, as it's just G-code.

Ghostkeeper, 2018-01-23:
This appears to be a bug in legacy Cura, so I'd like to refer you here: https://github.com/daid/LegacyCura/issues

Since that application hasn't updated since 15.04, something on your end must have changed.

ianpaschal, 2018-02-26:
Reporter has not provided more information for more than 2 weeks, so closing.

closed 2018-02-26 by ianpaschal