Ultimaker/Cura #3103: P

JPFrancoia, 2018-01-08:
**Application Version**
3.1.0-1

**Platform**
Archlinux

**Qt**
5.10.0-1

**PyQt**
5.9.2-2

**Steps to Reproduce**
Upgrade from Cura 3.0.3-1 to Cura 3.1.0-1 through package manager

**Actual Results**

Old profiles seem to be carried on, but some parameters were changed (line width, shell thickness, colling parameters, etc).

Manually importing backups of these profiles gives the same result.

**Expected results**

My profiles should be carried on, with the same parameters.
Importing profiles should give me expected parameters.

**Additional Information**

I updated Cura after my holidays. My old profiles seem to have been carried on, but some parameters were changed and I don't know why. I tried to import backups of these profiles, but each time the parameters were not exactly what they are supposed to be. For example, in one of my profiles the line width is supposed to be 0.6 mm. After update, it's now set to 0.57.

Why is that ?

I donwgraded to 3.0 now and importing these profiles works fine.

ChrisTerBeke, 2018-01-08:
We've seen this several times before, usually when the custom profiles had some errors in them. Could you upload your Cura user data directory so we can have a look? cc @LipuFei

JPFrancoia, 2018-01-08:
Here you are:

[cura_data.zip](https://github.com/Ultimaker/Cura/files/1611656/cura_data.zip)

LipuFei, 2018-01-08:
@JPFrancoia Thanks. I will have a look.

LipuFei, 2018-01-09:
@JPFrancoia I have made a few fixes in the master branch. Could you try it to see if your quality profiles are updated correctly?

JPFrancoia, 2018-01-10:
Ok I'll need a little help here.

I cloned the repo and tried tu run ```cura_app.py``` (it seemed a good idea)

```
cura.CrashHandler.__init__ [62]: An uncaught error has occurred!
cura.CrashHandler.__init__ [65]: Traceback (most recent call last):
cura.CrashHandler.__init__ [65]:   File "cura_app.py", line 94, in <module>
cura.CrashHandler.__init__ [65]:     if not cura.CuraApplication.CuraApplication.preStartUp(parser = parser, parsed_command_line = known_args):
cura.CrashHandler.__init__ [65]:   File "/home/djipey/Desktop/Cura/cura/CuraApplication.py", line 600, in preStartUp
cura.CrashHandler.__init__ [65]:     CuraApplication.addCommandLineOptions(parser, parsed_command_line = parsed_command_line)
cura.CrashHandler.__init__ [65]:   File "/home/djipey/Desktop/Cura/cura/CuraApplication.py", line 540, in addCommandLineOptions
cura.CrashHandler.__init__ [65]:     super().addCommandLineOptions(parser, parsed_command_line = parsed_command_line)
cura.CrashHandler.__init__ [65]: TypeError: addCommandLineOptions() got an unexpected keyword argument 'parsed_command_line'
Error in sys.excepthook:
Traceback (most recent call last):
  File "cura_app.py", line 75, in exceptHook
    _crash_handler.show()
  File "/home/djipey/Desktop/Cura/cura/CrashHandler.py", line 286, in show
    Application.getInstance().callLater(self._show)
  File "/usr/lib/python3.6/site-packages/UM/Application.py", line 321, in getInstance
    Application._instance = cls()
TypeError: __init__() missing 2 required positional arguments: 'name' and 'version'

Original exception was:
Traceback (most recent call last):
  File "cura_app.py", line 94, in <module>
    if not cura.CuraApplication.CuraApplication.preStartUp(parser = parser, parsed_command_line = known_args):
  File "/home/djipey/Desktop/Cura/cura/CuraApplication.py", line 600, in preStartUp
    CuraApplication.addCommandLineOptions(parser, parsed_command_line = parsed_command_line)
  File "/home/djipey/Desktop/Cura/cura/CuraApplication.py", line 540, in addCommandLineOptions
    super().addCommandLineOptions(parser, parsed_command_line = parsed_command_line)
TypeError: addCommandLineOptions() got an unexpected keyword argument 'parsed_command_line'
```

I'm not sure this is the right way. How do I run Cura from master ?

LipuFei, 2018-01-10:
@JPFrancoia I will create an AppImage for you with master branch.

JPFrancoia, 2018-01-10:
Thank you very much, I appreciate.

LipuFei, 2018-01-10:
@JPFrancoia There you go. https://send.firefox.com/download/39a83ac04d/#gas08yGVPzRUgYYs35ERKg

JPFrancoia, 2018-01-10:
Ok I just tested it. SOrry to say that, but it does absolutely nothing :/

I extracted it in my home and ran it. It started with the printer dialog box (none was present). None of my profile was there. I tried to import backed up profiles. I got a dialog box saying they were successfully imported, but they were not present in the combo box.

It seems this appImage is running in some sort of sandbox.

jackha, 2018-01-11:
@JPFrancoia can you add the logs from the first run? Maybe some errors pop up.

JPFrancoia, 2018-01-11:
Ok I understood the problem when I was looking for the logs, the app image created a 3.1 directory in my .local/share/cura directory. Here is a ls of .local/share/cura:

```
2.7  3.1  master  stderr.log  stdout.log
```

Cura installed from my package manager uses the master directory. Sorry for this little waste of time.

Now the serious business:

Machines were properly imported, but not the profiles. Especially this one:
[PP_local_transparent_0.6_nozzle.zip](https://github.com/Ultimaker/Cura/files/1622208/PP_local_transparent_0.6_nozzle.zip)

When I try to import this profile in the appImage, it says it imported correctly but it doesn't appear in the choices afterwards.

I don't know if it's a nozzle size problem (this profile is for an 0.6 nozzle). Also, I **absolutely hate** that the nozzle size is not part of the profile, and that it changes some profile parameters when I change the nozzle size.

Ghostkeeper, 2018-01-28:
Still reproduces in the 3.2 beta (on Windows). My reproduce steps were:

1. Factory reset (delete `%APPDATA%/Local/cura` and `%APPDATA%/Roaming/cura`).
2. Add Ultimaker 2+ machine on start-up.
3. Switch to PP, 0.6mm nozzle, Fast.
4. Download `PP_local_transparent_0.6_nozzle.zip` and rename it to `PP_local_transparent_0.6_nozzle.curaprofile`.
5. Import the profile in the profile manager.

It says the profile got successfully imported but it is not in the list. Here's my log:

    2018-01-28 15:28:14,536 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin FileLogger
    2018-01-28 15:28:14,579 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin FirmwareUpdateChecker
    2018-01-28 15:28:14,586 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin GCodeProfileReader
    2018-01-28 15:28:14,601 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin GCodeReader
    2018-01-28 15:28:14,608 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin GCodeWriter
    2018-01-28 15:28:14,619 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin ImageReader
    2018-01-28 15:28:14,627 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin LegacyProfileReader
    2018-01-28 15:28:14,634 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin LocalContainerProvider
    2018-01-28 15:28:14,645 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin LocalFileOutputDevice
    2018-01-28 15:28:14,654 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin MachineSettingsAction
    2018-01-28 15:28:14,662 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin MonitorStage
    2018-01-28 15:28:14,672 - WARNING - UM.Preferences.getValue [81]: Tried to get the value of non-existing setting cura/active_mode.
    2018-01-28 15:28:14,673 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin PerObjectSettingsTool
    2018-01-28 15:28:14,683 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin PluginBrowser
    2018-01-28 15:28:14,692 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin PostProcessingPlugin
    2018-01-28 15:28:14,699 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin PrepareStage
    2018-01-28 15:28:14,711 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin RemovableDriveOutputDevice
    2018-01-28 15:28:14,731 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin SimulationView
    2018-01-28 15:28:14,742 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin SliceInfoPlugin
    2018-01-28 15:28:14,751 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin SolidView
    2018-01-28 15:28:14,761 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin CameraTool
    2018-01-28 15:28:14,771 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin MirrorTool
    2018-01-28 15:28:14,783 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin RotateTool
    2018-01-28 15:28:14,795 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin ScaleTool
    2018-01-28 15:28:14,803 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin SelectionTool
    2018-01-28 15:28:14,815 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin TranslateTool
    2018-01-28 15:28:14,830 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin UltimakerMachineActions
    2018-01-28 15:28:14,887 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin UM3NetworkPrinting
    2018-01-28 15:28:14,900 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin UpdateChecker
    2018-01-28 15:28:14,902 - INFO - UpdateChecker.UpdateCheckerJob.run [34]: Checking for new version of cura
    2018-01-28 15:28:14,932 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin USBPrinting
    2018-01-28 15:28:14,940 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin UserAgreementPlugin
    2018-01-28 15:28:14,955 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade21to22
    2018-01-28 15:28:14,964 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade22to24
    2018-01-28 15:28:14,976 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade25to26
    2018-01-28 15:28:14,986 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade26to27
    2018-01-28 15:28:14,995 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade27to30
    2018-01-28 15:28:15,008 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade30to31
    2018-01-28 15:28:15,018 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin SimpleView
    2018-01-28 15:28:15,034 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin X3DReader
    2018-01-28 15:28:15,052 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin XmlMaterialProfile
    2018-01-28 15:28:15,063 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin XRayView
    2018-01-28 15:28:15,073 - INFO - UM.VersionUpgradeManager.upgrade [113]: Looking for old configuration files to upgrade.
    2018-01-28 15:28:15,080 - DEBUG - UM.Backend.Backend._logSocketState [173]: Socket state changed to Listening
    2018-01-28 15:28:15,084 - INFO - UM.Backend.Backend.startEngine [74]: Started engine process: C:\Program Files\Ultimaker Cura 3.2\CuraEngine.exe
    2018-01-28 15:28:15,085 - DEBUG - UM.Backend.Backend._backendLog [90]: [Backend] Calling engine with: ['C:\\Program Files\\Ultimaker Cura 3.2\\CuraEngine.exe', 'connect', '127.0.0.1:49674', '-j', 'C:\\Program Files\\Ultimaker Cura 3.2\\resou [... cut ...]
    2018-01-28 15:31:10,452 - DEBUG - CuraEngineBackend.CuraEngineBackend.slice [197]: starting to slice!
    2018-01-28 15:31:10,452 - DEBUG - CuraEngineBackend.CuraEngineBackend.slice [214]: Going to slice build plate [0]!
    2018-01-28 15:31:10,453 - DEBUG - CuraEngineBackend.CuraEngineBackend.slice [218]: Build plate 0 has no objects to be sliced, skipping
    2018-01-28 15:31:18,906 - DEBUG - AutoSave.AutoSave._onTimeout [48]: Autosaving preferences, instances and profiles
    2018-01-28 15:31:36,059 - DEBUG - AutoSave.AutoSave._onTimeout [48]: Autosaving preferences, instances and profiles

VMaxx, 2019-07-26:
This may be from creating a printer then renaming it.  I had someone recently report an error related to this.  They are getting the "Could not get metadata of container" and it was due to it being renamed, but the file did not.

Vandrasc, 2019-07-26:
It works for me when using the steps as described by @Ghostkeeper , i used the 4.2 Beta on win10. The profile is successfully imported and shown in the list.

GregValiant, 2024-11-14:
Is this still an issue or can this be closed?

closed 2024-12-10 by github-actions[bot]