Ultimaker/Cura #3104: "View print jobs" and "View printers" buttons fail on Linux with AppImage

sedwards2009, 2018-01-08:
**Application Version**
3.1.0 AppImage from ultimaker.com

**Platform**
Kubuntu 17.10

**Steps to Reproduce**
Simply clicking on the "View print jobs" or "View printers" buttons on the monitor tab when connected to a UM3.

**Actual Results**
Nothing. Web browser isn't opened.

**Expected results**
I expect the Cura Connect web page to appear in my browser.

**Additional Information**

The console output from the AppImage shows this:

```
kde-open5: ./bin/libQt5Core.so.5: version `Qt_5.9' not found (required by kde-open5)
kde-open5: ./bin/libQt5Core.so.5: version `Qt_5.9' not found (required by /usr/lib/x86_64-linux-gnu/libKF5KIOWidgets.so.5)
kde-open5: ./bin/libQt5Core.so.5: version `Qt_5.9' not found (required by /usr/lib/x86_64-linux-gnu/libKF5KIOCore.so.5)
kde-open5: ./bin/libQt5Core.so.5: version `Qt_5.9' not found (required by /usr/lib/x86_64-linux-gnu/libKF5I18n.so.5)
kde-open5: ./bin/libQt5Core.so.5: version `Qt_5.9' not found (required by /usr/lib/x86_64-linux-gnu/libKF5CoreAddons.so.5)
kde-open5: ./bin/libQt5Core.so.5: version `Qt_5.9' not found (required by /usr/lib/x86_64-linux-gnu/libKF5JobWidgets.so.5)
kde-open5: ./bin/libQt5Core.so.5: version `Qt_5.9' not found (required by /usr/lib/x86_64-linux-gnu/libKF5Service.so.5)
kde-open5: ./bin/libQt5Core.so.5: version `Qt_5.9' not found (required by /usr/lib/x86_64-linux-gnu/libKF5Completion.so.5)
kde-open5: ./bin/libQt5Core.so.5: version `Qt_5.9' not found (required by /usr/lib/x86_64-linux-gnu/libKF5IconThemes.so.5)
kde-open5: ./bin/libQt5Core.so.5: version `Qt_5.9' not found (required by /usr/lib/x86_64-linux-gnu/libKF5WindowSystem.so.5)
kde-open5: ./bin/libQt5Core.so.5: version `Qt_5.9' not found (required by /usr/lib/x86_64-linux-gnu/libKF5ConfigWidgets.so.5)
kde-open5: ./bin/libQt5Core.so.5: version `Qt_5.9' not found (required by /usr/lib/x86_64-linux-gnu/libKF5WidgetsAddons.so.5)
kde-open5: ./bin/libQt5Core.so.5: version `Qt_5.9' not found (required by /usr/lib/x86_64-linux-gnu/libKF5ConfigGui.so.5)
kde-open5: ./bin/libQt5Core.so.5: version `Qt_5.9' not found (required by /usr/lib/x86_64-linux-gnu/libKF5ConfigCore.so.5)
kde-open5: ./bin/libQt5Core.so.5: version `Qt_5.9' not found (required by /usr/lib/x86_64-linux-gnu/libQt5Xml.so.5)
kde-open5: ./bin/libQt5Core.so.5: version `Qt_5.9' not found (required by /usr/lib/x86_64-linux-gnu/libKF5DBusAddons.so.5)
kde-open5: ./bin/libQt5Core.so.5: version `Qt_5.9' not found (required by /usr/lib/x86_64-linux-gnu/libQt5X11Extras.so.5)
kde-open5: ./bin/libQt5Core.so.5: version `Qt_5.9' not found (required by /usr/lib/x86_64-linux-gnu/libKF5Archive.so.5)
kde-open5: ./bin/libQt5Core.so.5: version `Qt_5.9' not found (required by /usr/lib/x86_64-linux-gnu/libKF5ItemViews.so.5)
kde-open5: ./bin/libQt5Core.so.5: version `Qt_5.9' not found (required by /usr/lib/x86_64-linux-gnu/libKF5Codecs.so.5)
kde-open5: ./bin/libQt5Core.so.5: version `Qt_5.9' not found (required by /usr/lib/x86_64-linux-gnu/libKF5Auth.so.5)
kde-open5: ./bin/libQt5Core.so.5: version `Qt_5.9' not found (required by /usr/lib/x86_64-linux-gnu/libKF5GuiAddons.so.5)
```

My local KDE desktop does in fact use Qt 5.9. I'm guessing that Cura is running my local `kde-open5` command with the URL, but from the AppImage environment which has all of the bundled Cura specific Qt libraries in the path. `kde-open5` doesn't get along with Cura's Qt library version (nor should they even be getting mixed up in the first place).

ChrisTerBeke, 2018-01-08:
All the official builds use Qt 5.8, maybe it's a bug in 5.9?

sedwards2009, 2018-01-08:
My `kde-open5` command is built against Qt 5.9 and needs it, but it is being run in an environment (=Cura) which has Qt 5.8 libraries in there with higher prio. (`./bin/libQt5Core.so.5` is from the Cura AppImage.)

awhiemstra, 2018-01-09:
Yeah, the open command needs to be run with a clean environment. I also ran into this issue the other day, and there is another user who reported a similar issue on another DE, that I cannot find currently.

Ghostkeeper, 2018-01-25:
So either we need to run that command in the user's normal environment, or we'd need to include a version that does work inside the AppImage environment?

diegopradogesto, 2019-09-10:
The monitor tab has change a lot since then, and I think this is not happening anymore. The "Manage in browser" link works as expected.

closed 2019-09-10 by diegopradogesto