Ultimaker/Cura #3106: Start Gcode temperature variable incorrect value

mroote, 2018-01-08:
**Application Version**
3.1

**Platform**
Linux - Appimage

**Display Driver**
Nvidia 387

**Steps to Reproduce**
I have set the starting temperature to 200C and use this starting Gcode. For some reason the temperature becomes 210C and Cura adds its own temperature set value before this starting Gcode is ran.

```
M80                           ; turn power on
M117 Heating...               ;
M190 S{print_bed_temperature} ; start warming the bed wait for temp
M109 S{print_temperature}     ; Wait for extruder temp
G21                           ; metric values
G90                           ; absolute positioning
M117 Homing...                ;
M82                           ; set extruder to absolute mode
G28 X0 Y0                     ; home x+ y
G28 Z0                        ; home z
G1 Z5.0 F9000                 ; move the head 5mm up for CYA clearance
G92 E0                        ; zero the extruded length
M117 Priming...               ;
G1 X100 Y220 F4000            ; move half way along the back edge
G1 Z1                         ; move nozzle close to bed
G1 E15                        ; extrude 10 mm of filament
G1 Z10 F12000 E5              ; move up while extruding 10mm
G4 P2000                      ; wait 2 second for nozzle length to stabilize
G92 E0                        ; reset extruder
G1 X10 Y200 F4000             ; Move print head away from extruded filament
M117 Printing...              ;
```

**Actual Results**
Starting Gcode produced ends up like this with a duplicate temperature setting and the one from the custom starting script gets the value of 210C even though I have set everything to 200C.

```
     1 ;FLAVOR:Marlin                                                                  
     2 ;TIME:8674                                                                      
     3 ;Filament used: 7.75701m                                                        
     4 ;Layer height: 0.2                                                              
     5 ;Generated with Cura_SteamEngine 3.1.0                                          
     6 M190 S60                                                                        
     7 M104 S200                                                                       
     8 M109 S200                                                                       
     9 M82 ; absolute extrusion mode                                                   
    10 M80                           ; turn power on                                   
    11 M117 Heating...               ;                                                 
    12 M190 S60.0 ; start warming the bed wait for temp                                
    13 M109 S210     ; Wait for extruder temp                                          
    14 G21                           ; metric values                                   
    15 G90                           ; absolute positioning                            
    16 M117 Homing...                ;                                                 
    17 M82                           ; set extruder to absolute mode                   
    18 G28 X0 Y0                     ; home x+ y                                       
    19 G28 Z0                        ; home z                                          
    20 G1 Z5.0 F9000                 ; move the head 5mm up for CYA clearance          
    21 G92 E0                        ; zero the extruded length                        
    22 M117 Priming...               ;                                                 
    23 G1 X100 Y220 F4000            ; move half way along the back edge               
    24 G1 Z1                         ; move nozzle close to bed                        
    25 G1 E15                        ; extrude 10 mm of filament                       
    26 G1 Z10 F12000 E5              ; move up while extruding 10mm                    
    27 G4 P2000                      ; wait 2 second for nozzle length to stabilize    
    28 G92 E0                        ; reset extruder                                  
    29 G1 X10 Y200 F4000             ; Move print head away from extruded filament     
    30 M117 Printing...              ;  
```

**Expected results**
I would expect Cura to drop the initial temperature set value.  Also I'm not sure where the 210 is coming from since I have set all temperatures to be 200C.

Do I have the correct variables in my starting script? Also is there a list somewhere with all the possible values for the script? Thanks

ChrisTerBeke, 2018-01-08:
Duplicate of #3061, #3032, #1296 and others. Will be fixed in Cura 3.2

closed 2018-01-08 by ChrisTerBeke

mroote, 2018-01-08:
When is 3.2 expected to be released?

ChrisTerBeke, 2018-01-08:
Beta is expected in 2 weeks, stable 2 weeks after that. I don't recommend using the Beta for any production work.

mroote, 2018-01-08:
Great thanks!